neuropsychology/NeuroKit #602: RPeaks are calculated twice in ecg_process

CReizner, 2022-01-10:
Within the function **ecg_process**, you are calculating rpeaks with **ecg_peaks** but do not pass it to the **ecg_quality** function and **_ecg_quality_averageQRS** will therefore calculate the rpeaks list again. 

```
def ecg_process(ecg_signal, sampling_rate=1000, method="neurokit"):
    ...
    instant_peaks, rpeaks, = ecg_peaks(
        ecg_cleaned=ecg_cleaned, sampling_rate=sampling_rate, method=method, correct_artifacts=True
    )
    ...
    quality = ecg_quality(ecg_cleaned, !!!rpeaks=None!!!, sampling_rate=sampling_rate)
    ...
```

zen-juen, 2022-01-11:
Hi @CReizner thanks for catching this! Indeed you're right, it should be `rpeaks=rpeaks` rather than `rpeaks=None` in `ecg_process` to avoid peak computation again. Would you like to open a [pull request](https://github.com/neuropsychology/NeuroKit/compare) to the dev branch here?

CReizner, 2022-01-11:
Hi @zen-juen. It should be  ```rpeaks=rpeaks['ECG_R_Peaks']```, because it wants a list. I honestly don't know how git or github works, so I don't see myself creating a pull request =D

DominiqueMakowski, 2022-01-11:
No worries, we'll fix it then. Thanks a lot for letting us know!

PR #691 (2022-08-18): [Fix] Added detected rpeaks to ecg_process
# Description

This PR fixes #602 . It makes such that the detected RPeaks get added to `ecg_quality`.

Runs in `docs/examples/bio_custom/bio_custom.ipynb`. May have to adapt tests.
# Proposed Changes

- `rpeaks["ECG_R_Peaks"]` is used instead of `None` for `ecq_quality`.
- Some automatic formatting

# Checklist

- [x] I have read the [CONTRIBUTING](https://github.com/neuropsychology/NeuroKit/blob/master/.github/CONTRIBUTING.rst#structure-and-code) file.
- [x] My PR is targetted at the **dev branch** (and not towards the master branch).
- [x] I ran th [... cut]
closed 2022-09-27 by stale[bot]