neuropsychology/NeuroKit #603: Error for EDA data

MostafaM2020, 2022-01-11:
Hi all,

Sampling rate for my data is 2 Hz and I used the solution presented in issue #554 to solve the low sampling problem. However, I'm receiving following error for some part of my data when I use `eda_peaks()` function or when I resample the data. Can you please let me know how can I solve the problem? You can find in the following link some data which have this issue with.

`ValueError: zero-size array to reduction operation maximum which has no identity`

[Book1.xlsx](https://github.com/neuropsychology/NeuroKit/files/7843500/Book1.xlsx)

Thank you for your help in advance.

zen-juen, 2022-01-11:
Hi @MostafaM2020 could you provide us the with an example script so that we can reproduce the error and investigate? Thanks :)

MostafaM2020, 2022-01-11:
Thank you @zen-juen . Following is the used script.

```
def eda_custom_process(eda_signal, sampling_rate=2, method="neurokit"):
    

    eda_signal = nk.signal_sanitize(eda_signal)
    
    # Series check for non-default index
    if type(eda_signal) is pd.Series and type(eda_signal.index) != pd.RangeIndex:
        eda_signal = eda_signal.reset_index(drop=True)
    
    # Preprocess
    eda_cleaned = eda_signal  #Add your custom cleaning module here or skip cleaning
    eda_decomposed = nk.eda_phasic(eda_cleaned, sampling_rate=sampling_rate)

    # Find peaks
    peak_signal, info = nk.eda_peaks(
        eda_decomposed["EDA_Phasic"].values,
        sampling_rate=sampling_rate,
        method=method,
        amplitude_min=0.1,
    )
    info['sampling_rate'] = sampling_rate  # Add sampling rate in dict info

    # Store
    signals = pd.DataFrame({"EDA_Raw": eda_signal, "EDA_Clean": eda_cleaned})

    signals = pd.concat([signals, eda_decomposed, peak_signal], axis=1)

    return signals, info

df = pd.read_excel('Book1.xlsx')
data= df['Data 1']
signals, info = eda_custom_process(data, sampling_rate=2)
```

zen-juen, 2022-01-11:
I was able to run this quite smoothly!

Did you then encounter the above error when running `nk.eda_peaks(signals["EDA_Phasic"], sampling_rate=2)`?

MostafaM2020, 2022-01-11:
Thank you @zen-juen for your help. 
That's interesting. It still gives me the error when I import the data from the main files where I saved them. But, I'm not receiving the error once I copy the data to a new excel file. I have no idea why it's happening.

closed 2022-01-11 by MostafaM2020