neuropsychology/NeuroKit #608: Chart X-Axis and Titles Overlapping in Colab

windowshopr, 2022-01-18:
[image]

Any ideas?? :)

anshu-97, 2022-01-19:
Hi @windowshopr, while there's a few ways of going about this, the more straightforward method would probably be to add a line to at the end to call the `plt.subplots_adjust()` function, which allows you to adjust the positions of the subplots within the figure by specifying the values of the arguments `wspace` (width of space between subplots) and `hspace` (height of space between subplots).

I have provided an example of how my code and generated plot looks like below:
[image]

Hopefully this works for you!

windowshopr, 2022-01-19:
Thanks!!!! Worked for me!