neuropsychology/NeuroKit #609: Filters do not compute the initial conditions

mscheltienne, 2022-01-19:
I started using neurokit for a paradigm exploiting the ECG signal in real-time. I want to use `'hamilton2002'` method. As I am working on a real-time paradigm, data is coming by chunks in a buffer initialized to 0. Thus, when filtering, the filter response introduces a massive artifact at the beginning of all my windows because of the DC component. 

(On the left, one of the first buffers where you still see the DC step; On the right, a buffer a few seconds later)

[image]

https://github.com/neuropsychology/NeuroKit/blob/d01111b9b82364d28da01c002e6cbfc45d9493d9/neurokit2/ecg/ecg_clean.py#L200-L209

I see that you are using everywhere `ba` output and that you are not computing the initial conditions. I propose to replace the output `ba` with `sos` that is more numerically stable and recommended and to compute the initial conditions with [`sosfilt_zi`](https://docs.scipy.org/doc/scipy/reference/generated/scipy.signal.sosfilt_zi.html?highlight=filt_zi#scipy.signal.sosfilt_zi).

DominiqueMakowski, 2022-01-19:
That seems reasonable. I think the reason why it's a `ba` filter was to follow the original implementation or description; but I do agree that in this case the algorithm doesn't rely on the type of filtering arguments, so we could probably use the most appropriate one.

Would you mind contributing a PR to fix that ☺️ ?

mscheltienne, 2022-01-19:
I'll open a PR this week if I can, or in the beginning of February.

DominiqueMakowski, 2022-01-19:
Awesome, looking forward to your collab'

PR #612 (2022-02-12): Replace ba output with sos for ECG filters and determine initial conditions
Closes #609

I replaced `ba` output with `sos` when needed, and I added the initial condition `zi` if `filt` was used instead of `filtfilt`.
Even if you want to replicate as closely as possible a method, an implementation, IMO, using `sos` instead of `ba` should always be preferred as it should not induce significant differences while being more numerically stable. 

I didn't see I had to branch from `dev` and not `master`; let's see if any conflicts arise.
Also, I added `.DS_Store` to the .gitignore for macOS develope [... cut]
closed 2022-02-14 by mscheltienne