neuropsychology/NeuroKit #610: ECG neurokit cleaning method filters powerline at 50 Hz

mscheltienne, 2022-01-19:
https://github.com/neuropsychology/NeuroKit/blob/d01111b9b82364d28da01c002e6cbfc45d9493d9/neurokit2/ecg/ecg_clean.py#L129

Powerline noise is at 60 Hz in the US. Maybe have the `**kwargs` passed to the private functions? But that would require some documentation on a per-method basis.. Or maybe just get rid of powerline filtering here and replace with a BP at e.g. 40 or 45 Hz cutoff?

DominiqueMakowski, 2022-01-19:
mmh I think it's worth adding this argument (as an exposed one) for the cleaning functions since it's something that you would expect to potentially tweak at this stage. And we can pass it to ecg_clean as kwargs from higher level functions like ecg_process? 

Thanks a lot for the suggestions, again would you like / have time to make this argument tweakable in ecg_clean in a PR?

mscheltienne, 2022-01-20:
I could add this to #612; but I don't exactly get what you propose.
It does not make sense to me to expose a `powerline` argument to `ecg_clean` since it's used only in one of the cleaning methods.

mscheltienne, 2022-08-01:
Up to prevent closure as this was probably not fixed?

mscheltienne, 2022-10-09:
Looks like it was fixed as now the `kwargs` are forwarded https://github.com/neuropsychology/NeuroKit/blob/5cd089e9d82e6f7fb98d2028e69840aee08f8ab3/neurokit2/ecg/ecg_clean.py#L147-L155

closed 2022-10-09 by mscheltienne