neuropsychology/NeuroKit #611: Regarding the ppg_clean() function

CrazyPrincipal, 2022-01-20:
**Regarding the ppg_clean() function.**
PPG_cleaned = nk.ppg_clean ( ppg_signal , sampling_rate = 250 , method = 'elgendi' ) 
After executing this code, as shown below, there is a huge spike at the beginning and end, which is not present in the original signal, so that the subsequent code does not work.

I can't figure out what's wrong with the pgg_clean() function, so please help me out.
And attach the original signal "PPG_20220119-184610.xlsx".

Thank you very much!

[image]
[PPG_20220119-184610.xlsx](https://github.com/neuropsychology/NeuroKit/files/7902459/PPG_20220119-184610.xlsx)

CrazyPrincipal, 2022-01-20:
> **Regarding the ppg_clean() function.** PPG_cleaned = nk.ppg_clean ( ppg_signal , sampling_rate = 250 , method = 'elgendi' ) After executing this code, as shown below, there is a huge spike at the [...]

anshu-97, 2022-01-20:
Hi @CrazyPrincipal, the `ppg_clean(method='elgendi')` function makes use of the lower order `signal_filter` function. This function uses the Butterworth filter and passes the [ `ba` output ](https://docs.scipy.org/doc/scipy/reference/generated/scipy.signal.butter.html?highlight=butter#scipy.signal.butter) to filter the signal https://github.com/neuropsychology/NeuroKit/blob/d01111b9b82364d28da01c002e6cbfc45d9493d9/neurokit2/ppg/ppg_clean.py#L98-L103. 
However, numerical problems can occur since the conversion between roots and the polynomial coefficients is a numerically sensitive operation. I am guessing that this might by why your cleaned ppg signal looks incorrect. This can be fixed by tweaking the code to replace the `ba` output with the `sos` output (which is more numerically stable).

Alternatively, you could also choose to clean your signal using Nabian et al (2018)'s  method of filtering instead. I have attached an image of the code I used and the image of the plots using `ppg_clean(method='nabian2018')` and `sos` output below. 

[image]

[image]

You can also read more about the Butterworth filter here if you are interested!
https://docs.scipy.org/doc/scipy/reference/generated/scipy.signal.butter.html?highlight=butter#scipy.signal.butter

Hope this helps :)

CrazyPrincipal, 2022-01-21:
> 嗨@CrazyPrincipal，该`ppg_clean(method='elgendi')`函数使用了低阶`signal_filter`函数。该函数使用巴特沃斯滤波器并通过[输出](https://docs.scipy.org/doc/scipy/reference/generated/scipy.signal.butter.html?highlight=butter#scipy.signal.butter)对信号进行滤波[ `ba`](https://docs.scipy.org/doc/scipy/reference/generated/scipy.signal.butter.html?highlight=butter#scipy.signal.butter)
> 
> https://github.com/neuropsychology/NeuroKit/blob/d01111b9b82364d28da01c002e6cbfc45d9493d9/neurokit2/ppg/ppg_clean.py#L98-L103
> 
> .
> 然而，由于根和多项式系数之间的转换是对数值敏感的操作，因此可能会出现数值问题。我猜这可能是您清理后的 ppg 信号看起来不正确的原因。这可以通过调整代码以将`ba`输出替换为`sos`输出（在数值上更稳定）来解决。
> 或者，您也可以选择使用 Nabian et al (2018) 的过滤方法来清理您的信号。我附上了我使用的代码的图像以及下面使用`ppg_clean(method='nabian2018')`和`sos`输出的绘图图像。
> 
> [image]
> 
> [image]
> 
> 如果您有兴趣，还可以在此处阅读有关 Butterworth 过滤器的更多信息！ https://docs.scipy.org/doc/scipy/reference/generated/scipy.signal.butter.html?highlight=butter#scipy.signal.butter
> 
> 希望这可以帮助 ：）

Thank you very much for your help!

I have successfully solved it.
Likewise, I encounter this problem in the PPG_Clean data returned in the ppg_process() function, and I'm trying to solve it myself.

Thank you very much!

CrazyPrincipal, 2022-01-21:
@anshu-97

CrazyPrincipal, 2022-01-21:
@anshu-97 

Thank you very much for your help!

I have successfully solved it.
Likewise, I encounter this problem in the PPG_Clean data returned in the ppg_process() function, and I'm trying to solve it myself.

Thank you very much!

anshu-97, 2022-01-21:
Hi @CrazyPrincipal, 

Glad that this worked for you! I will go ahead and close this issue then but do feel free to open a separate issue referencing this one again (#611) if you continue to have problems using the `ppg_process()` function!

closed 2022-01-21 by anshu-97