neuropsychology/NeuroKit #613: Error running ecg_process and ecg_analyze examples (TypeError: unsupported operand type(s) for /: 'str' and 'int')

lauramhamm, 2022-01-23:
I receive the following error:

TypeError: unsupported operand type(s) for /: 'str' and 'int'

when running the ecg_process and ecg_analyze examples from https://neurokit2.readthedocs.io/en/latest/functions.html#module-neurokit2.ecg

For example, the following returns the unsupported operand error:

>>> import neurokit2 as nk
>>>
>>> ecg = nk.ecg_simulate(duration=15, sampling_rate=1000, heart_rate=80)
>>> signals, info = nk.ecg_process(ecg, sampling_rate=1000)

I know that the example ecg data are float types, and manually converting the dataframe values to float does not change the outcome.

System Specifications: 
- OS: Windows (WindowsPE 64bit)
- Python: 3.10.2
- NeuroKit2: 0.1.5

- NumPy: 1.22.1
- Pandas: 1.4.0
- SciPy: 1.7.3
- sklearn: 1.0.2
- matplotlib: 3.5.1

zillekem, 2022-01-31:
Try downgrading pandas to 1.3.5.

With pandas 1.4.0 the following code has a dtype of _object_ instead of _float64_ for `data["Signal"]`:

```python
buffer = pd.DataFrame(index=range(length_buffer), columns=data.columns)
data = data.append(buffer, ignore_index=True, sort=False)
```
_epochs_create.py, line 127-128_

closed 2022-02-15 by DominiqueMakowski

Manthan-tech, 2022-06-06:
I am facing the same issue, with the latest libraries. Is it fixed? What is the solution

DominiqueMakowski, 2022-06-07:
Does it still happen with the latest (> 0.2.0) version of NK?

commit ff5412fe (2022-02-09): fix #613
commit 0cf889d0 (2022-02-09): FAILED ATTEMPT TO FIX #613
commit 4194fede (2022-02-10): Finally fix #613