neuropsychology/NeuroKit #615: Minimum, maximum & percentiles as HRV features

danibene, 2022-02-04:
**Describe the solution you'd like**
Minimum, maximum, 20th percentile, and 80th percentile NN are sometimes included as time-domain HRV features, e.g.

- Minimum & maximum: [Subramaniam, S. D., & Dass, B. (2022). An Efficient Convolutional Neural Network for Acute Pain Recognition Using HRV Features. In Proceedings of the International e-Conference on Intelligent Systems and Signal Processing (pp. 119-132). Springer, Singapore.](https://doi.org/10.1007/978-981-16-2123-9_9)
- 20th & 80th percentile: [Han, L., Zhang, Q., Chen, X., Zhan, Q., Yang, T., & Zhao, Z. (2017). Detecting work-related stress with a wearable device. Computers in Industry, 90, 42-49.](https://doi.org/10.1016/j.compind.2017.05.004)

**How could we do it?**
Here is an example of a modified hrv_time function: https://github.com/danibene/NeuroKit/commit/bef5b8a136a00c931233e9fa5dd051a802d902b3

Would you be interested in this?

DominiqueMakowski, 2022-03-25:
Hi @danibene, sorry for the delay in responding we've had some changes in the team that slowed the maintenance of the package but we're slowly getting back at it :)

It looks like a good idea, would you be interested in contributing these changes by making a PR?

danibene, 2022-03-25:
No problem at all! 

I'd be happy to make a PR, but I'm a bit stuck at the [code checks](https://github.com/neuropsychology/NeuroKit/blob/master/.github/CONTRIBUTING.rst#run-code-checks). 

After changing my directory to the "hrv" folder, I tried running the following command to test the hrv_time script  that I modified:
```
pylint hrv_time.py --max-line-length=127 --load-plugins=pylint.extensions.docparams --load-plugins=pylint.extensions.docstyle --variable-naming-style=any --argument-naming-style=any --reports=n --suggestion-mode=y --disable=E303 --disable=R0913 --disable=R0801 --disable=C0114 --disable=E203 --disable=E0401 --disable=W9006 --disable=C0330 --disable=R0914 --disable=R0912 --disable=R0915 --disable=W0102 --disable=W0511 --disable=C1801 --disable=C0111 --disable=R1705 --disable=R1720 --disable=C0301 --disable=C0415 --disable=C0103 --disable=C0302 --disable=R1716 --disable=W0632 --disable=E1136 --extension-pkg-whitelist=numpy
```
This resulted in the following output:
```
************* Module hrv_time
hrv_time.py:7:0: E0402: Attempted relative import beyond top-level package (relative-beyond-top-level)
hrv_time.py:8:0: E0402: Attempted relative import beyond top-level package (relative-beyond-top-level)
hrv_time.py:11:0: W9015: "kwargs" missing in parameter documentation (missing-param-doc)

-----------------------------------
Your code has been rated at 8.98/10
```

I'm confused about how I should address these errors because I don't think my modifications resulted in them - but maybe I'm wrong. This is my first time trying to contribute to an open-source package like this, so I'd really appreciate any guidance. 

Here is the up-to-date branch in case helpful: https://github.com/danibene/NeuroKit/tree/feature/min_max_perc_hrv

Thanks so much :)

DominiqueMakowski, 2022-03-26:
worst case scenario I'll commit the changes directly, but I believe it's good if give it another try (I know github can be frustrating at first)

What I would suggest is:

- delete your fork and all (start fresh)
- fork the repo again
- put yourself on the "dev" branch
- make a minor change
- directly try to open a PR (i.e., to connect your fork to the main one, so that I can see it and directly commit/comment/review) (don't bother about checks / tests locally)
- you can continue editing and committing to that PR after it's opened

danibene, 2022-03-26:
I'd prefer to avoid deleting the fork if possible (I've made various other changes on a different branch for a project), so I tried comparing the code check output for the dev branch and the min_max_perc_hrv branch.

```
pylint hrv_time.py --max-line-length=127 --load-plugins=pylint.extensions.docparams --load-plugins=pylint.extensions.docstyle --variable-naming-style=any --argument-naming-style=any --reports=n --suggestion-mode=y --disable=E303 --disable=R0913 --disable=R0801 --disable=C0114 --disable=E203 --disable=E0401 --disable=W9006 --disable=C0330 --disable=R0914 --disable=R0912 --disable=R0915 --disable=W0102 --disable=W0511 --disable=C1801 --disable=C0111 --disable=R1705 --disable=R1720 --disable=C0301 --disable=C0415 --disable=C0103 --disable=C0302 --disable=R1716 --disable=W0632 --disable=E1136 --extension-pkg-whitelist=numpy
```

I had the following output for both the dev branch and the one with my changes:

```
************* Module neurokit2.hrv.hrv_time
hrv_time.py:11:0: W9015: "**kwargs" missing in parameter documentation (missing-param-doc)
hrv_time.py:138:20: C0201: Consider iterating the dictionary directly instead of calling .keys() (consider-iterating-dictionary)

-----------------------------------
Your code has been rated at 9.81/10
```

Not sure why the output is different from yesterday - maybe since I'm using a different machine today, but I pulled the changes I made yesterday...

In any case, neither of these warnings/recommendations are about lines I changed, so I went ahead and opened the PR: https://github.com/neuropsychology/NeuroKit/pull/627

closed 2022-03-29 by danibene