neuropsychology/NeuroKit #618: event_find method error on

deboracerretini, 2022-02-16:
I used the **event_find()** method using the _start_at_ parameter, like this:

        events = nk.events_find(myData, threshold_keep='above', start_at=1)

and I got this error:

          File "C:\Users\Utente\AppData\Local\Programs\Python\Python39\lib\site-packages\neurokit2\events\events_find.py", line 109, in events_find
    events["duration"] = events["duration"][events["onset"] >= start_at]
IndexError: boolean index did not match indexed array along dimension 0; dimension is 5 but corresponding boolean dimension is 4

I think that the problem is in this part of code of event_find.py (from 106 to 112):

    # Remove based on index #start
    if start_at > 0:
        events["onset"] = events["onset"][events["onset"] >= start_at] #row a
        events["duration"] = events["duration"][events["onset"] >= start_at] #row b
    if end_at is not None:
        events["onset"] = events["onset"][events["onset"] <= end_at] 
        events["duration"] = events["duration"][events["onset"] <= end_at]

I printed successfully the following BEFORE start (before row 106):

_events["onset"] [   0  191  609 1040 1465]
events["duration"] [ 170  397  410  405 1241]_

event["onset"] is used to discriminate the right and wrong events detected, but it is updated before it finished its purpose. That is to say, it is updated in row a and then used in row b, but to be correct it showld be used first and updated later.
In my case, Inverting row a and row b solved the problem.

The same thing can be thought for line 111 and 112.

Am I getting something wrong? Did I used the event_find method in a wrong way?

I'm sorry for my bad English, I hope it's clear what I meant.

I'm using:

- Python 3.9.9
- neurokit2 Version: 0.1.6

Thank you for the attention.

anshu-97, 2022-02-17:
Hi @deboracerretini, could you provide a sample of your data so that we can try to reproduce the error for investigation? 😄

deboracerretini, 2022-02-17:
mydata has
- type <class 'numpy.ndarray'>
- dtype float64

and contains the values in the file attached.
[values.txt](https://github.com/neuropsychology/NeuroKit/files/8088426/values.txt)

I hope it's clear :)

anshu-97, 2022-02-18:
Hi @deboracerretini, you are quite right in your explanation! Indeed because `events['onset']= events["onset"][events["onset"]>= start_at]` would have already filtered out the values before `'start_at'`, there would be a mismatch in the index between `[events["onset"]]` and `[events["duration]]` such that `events["duration"][events["onset"]` wouldn't be able to slice out what is indicated, causing the error you got. 

Thanks for catching that! 

We'll make amendments to the original code with your suggested correction 😄

deboracerretini, 2022-02-18:
Hi @anshu-97, right now that's all I spotted, thank you for checking!

closed 2022-02-21 by anshu-97