neuropsychology/NeuroKit #620: False positive R peaks detected with neurokit method : T onsets detected in addition to R peaks

Jean-Diddy, 2022-02-18:
Hello, 
Thank you for this amazing package ! 
I am trying to detect the R peaks with ECG data using ecg_findpeaks(). However it seems to detect a lot of T peaks with the current parameters set in the neurokit's method as shown on the graphs attached.

Is there a way I can I tune the parameters from of the neurokit method to solve this issue ?

[image]
[image]
[image]

Best,

Rudy

JanCBrammer, 2022-02-19:
@Jean-Diddy, thanks for the issue. Have a look at the following issues regarding the parameters of the `neurokit` peak detection: https://github.com/neuropsychology/NeuroKit/issues/321#issuecomment-656005790, https://github.com/JanCBrammer/biopeaks/issues/23#issuecomment-875514227. Unfortunately, you cannot tune the parameters of `_ecg_findpeaks_neurokit()` from `ecg_findpeaks()`. However, you can call `_ecg_findpeaks_neurokit()` on your signal directly.

Alternatively, you could try calling `ecg_findpeaks()` with different methods. The T-waves in your signal are quite large, maybe you find a method that can handle this better than `neurokit`?

DominiqueMakowski, 2022-02-19:
Actually since 1.6 you should be able to pass any additional arguments to `ecg_findpeaks()` that should get passed down to `ecg_findpeaks_neurokit`, such as these ones:

https://github.com/neuropsychology/NeuroKit/blob/a30f76e64b4108abdc652a20391dc0288c62501d/neurokit2/ecg/ecg_findpeaks.py#L279-L288

However, there are undocumented. Perhaps we should add a bit of documentation for them somewhere

Jean-Diddy, 2022-02-20:
@JanCBrammer @DominiqueMakowski Thank you for your answers. 
First of all, the QRS detection based methods seems to perform somehow better, and the Nabian 2018 one is performing better.

However I was wondering if adding a condition on the sign of the gradient at the position of the peak detected could help to decrease the False positive ratio as it is more like that an R peak occurs on a positive gradient (and the T-waves peaks on a negative gradient) of the smoothing signal. Yet, sorry, it may be a bad idea as I may miss some important points of the Neurokit algorithm peak detection...

@DominiqueMakowski I tried to pass these arguments on the ecg_findpeaks() like this : 
`kwargs = {'smoothwindow' : 0.2,
               'avgwindow' : 0.75,
               'gradthreshweight' : 1.5,
               'minlenweight' : 0.4,
               'mindelay' : 0.3}
    
    ecg_info = nk.ecg_findpeaks(ecg_cleaned, sampling_rate=fs_ecg, method='neurokit', show = True, **kwargs)`

But I get an issue telling me that ecg_findpeaks doesn't have an arg named "smoothwindow". I must miss a something using kwargs, right ?

About the documentation on the use of these arguments, it can be great, but I can play with the values and read the source code to understand it. Then I can give you feedback about what's work better for data I used

DominiqueMakowski, 2022-02-21:
Kwargs just mean "additional keywords arguments" (i.e., arguments with names), you need to pass them directly like so:

```python
ecg_info = nk.ecg_findpeaks(ecg_cleaned, sampling_rate=fs_ecg, method='neurokit', 
                            show = True, avgwindow = 0.75, smoothwindow = 0.2)` 
```

JanCBrammer, 2022-02-21:
> However I was wondering if adding a condition on the sign of the gradient at the position of the peak detected could help to decrease the False positive ratio as it is more like that an R peak occurs on a positive gradient (and the T-waves peaks on a negative gradient) of the smoothing signal.

It's an interesting observation. However, I'm not sure if this rule would still be relevant in other datasets with other ECG characterictics. It might be that this rule is overfit to your particular dataset, since the T-waves in your ECG are very prominent. 

Would be interesting to explore if/how adding this rule would change the detector's performance on benchmark datasets with diverse ECG characteristics! I won't have time in the forseeable future to dive into this, however, @Jean-Diddy, if you're interested in setting up a small experiment I'd be happy to answer your questions should you have some.

Jean-Diddy, 2022-02-25:
Thank you for your help @DominiqueMakowski and @JanCBrammer ! I was still in the 1.5 neurokit2 version that's why I couldn't change the args...sorry for this.

I don't have the time right now to test this, but I have around 150 ecg signals from 50 different people of 30 minutes high quality recordings that I processed with a smoothing window of 0.2 instead of 0.1 and the T peaks were not detected anymore without decreasing the sensitivity for R peaks' detection. This solution makes sense as the T-peaks are more likely to be under the smoothing signal as the influence of the R peaks become stronger over timestamps. 

However, I don't know to what extent it can be applied to PPG signals, but intuitively it shouldn't change the accuracy for PPG peak detection with smoothwindow = 0.2.

For the gradient sign idea, I didn't tested it yet as the issue was solved by increasing the smoothing window length for neurokit algorithm. But if the higher the T-waves, the less accurate the gradient condition I thought about would work, because the gradient of the smoothing window is less likely to be negative on the T-peak in this case. So if the T peaks or P peaks are much larger than R peaks (which is very unlikely to occur I guess, unless R peaks are reversed), then the gradient condition will decrease the sensitivity of R peak detection.

I will contact you again when I have time to conduct a small experiment to kind of benchmark objectively  this new parameter or solution with your dataset :) 

Thank you again for your help and this package and sorry, my english is not perfect, I just hope it is clear enough :)

Jean-Diddy, 2022-03-09:
@JanCBrammer Unfortunately, the database is proprietary for the moment, yet I totally understand the need for benchmarking and the value it has.

Ok, thanks a lot !

closed 2022-03-09 by Jean-Diddy