neuropsychology/NeuroKit #626: MNE 1.0: adjustments

DominiqueMakowski, 2022-03-23:
```python
import neurokit2 as nk
import mne

raw = mne.io.read_raw_fif(
        mne.datasets.sample.data_path() + "/MEG/sample/sample_audvis_raw.fif", preload=True
    )
x = nk.mne_channel_extract(raw, what="EOG", name="EOG")
x
```

This used to return a vector (I think?), but now it returns a series (at least based on the bug in the test that caught this)

TODO: need to investigate `mne_channel_extract()` and see if changes are to be made based on the most recent MNE update

JoseAlanis, 2022-10-10:
Hi, is this still an issue, I could take a look 😃

DominiqueMakowski, 2022-10-11:
I "think" it has been solved? since we don't have tests failing at least... But if they are other MNE-related things (and there will be, as enhancing the M/EEG aspect of NK is one of the goals), I'll let you know :)

btw, great job on the MNE-LIMO toolbox

closed 2022-10-11 by DominiqueMakowski