neuropsychology/NeuroKit #628: hrv_frequency() use different frequency bandings

RichardBJ, 2022-03-28:
Hi, I noticed that, for me, when I tried to use different frequency bandings (ulf=(x,y) etc), these did not seem to propagate through to modified bandings on my periodograms.

I have forked the NeuroKit repository and fixed that now by edits to the hrv_frequency.py here
[https://github.com/RichardBJ/NeuroKit](url)

Would you like to incorporate this?  If so just let me know.

ps., wonderful package.  I was actually mid way through converting the Fortran ECGPUWAVE program into Python when someone alerted me to Neurokit, so I abandoned that approach now and use, with great appreciation your ecg_delineate() function!! Thanks so much for all this work.
Richard

RichardBJ, 2022-03-28:
Thanks @DominiqueMakowski, happy to do that.  Will need to read the instructions first :-)  but good practise!

luiseduve, 2022-07-15:
Wasn't this issue solved in #630?

DominiqueMakowski, 2022-07-15:
Good catch @luiseduve  thanks 😁

closed 2022-07-15 by DominiqueMakowski