neuropsychology/NeuroKit #650: Adapt signal_fixpeaks to deal with larger gaps in data

danibene, 2022-06-03:
**Describe the solution you'd like**
I noticed that the "neurokit" method of signal_fixpeaks does not work well with larger gaps in the data, sometimes leading to negative indices or getting stuck in a loop: https://github.com/danibene/fixpeaks_with_large_gaps/blob/main/example_signal_with_large_gaps.ipynb

**How could we do it?**
I think there are a few possible ways to go about dealing with larger gaps (not mutually exclusive):
1. Adapting the number of NaNs depending on the interval size rather than always inserting 2 NaNs for large intervals
2. Interpolating the peaks directly rather than the intervals
3. Providing a warning if there are negative indices & breaking the loop after a certain number of iterations

If you're interested in any of the above, I'm happy to try my hand & open a PR!

DominiqueMakowski, 2022-06-03:
Fixing peaks is a tricky issue, but there's definitely room for improvement!
n°3 should be a starting point, then I don't know about what's best between 1 and 2... but please do open a PR so we can check. 

Would be also nice to add an example / study explaining and benchmarking these methods

danibene, 2022-06-06:
> Fixing peaks is a tricky issue
Indeed!

I've opened the PR for now with the 3 changes I mentioned above, it's a work in progress but I thought I should get your input before overcomplicating things: https://github.com/neuropsychology/NeuroKit/pull/647 

[The new version is able to fix ECG peaks](https://github.com/danibene/fixpeaks_with_large_gaps/blob/main/example_ECG_signal_distored_nk_mod.ipynb 
) for data [where the old version got stuck in a loop](https://github.com/danibene/fixpeaks_with_large_gaps/blob/main/example_ECG_signal_distored_nk_orig.ipynb).

I still see a couple of issues though:
- there is no extrapolation in case there is a NaN at the final index (so when interpolating the peaks directly, the [index stays constant rather than increasing](https://github.com/danibene/fixpeaks_with_large_gaps/blob/main/example_signal_with_NaN_at_end.ipynb)) 
&rarr; modify the signal_interpolate function to extrapolate? 
&rarr; could remove NaNs at the end?
- sometimes the indices are not negative, but still [out of order](https://github.com/danibene/fixpeaks_with_large_gaps/blob/main/example_signal_with_NaN_at_end.ipynb)
&rarr; provide warning for peak indices that are not increasing? 
- if there are many missing peaks, it might be better to exclude them instead of linearly interpolating for certain analyses, e.g. short term HRV
&rarr; have option to keep NaNs rather than interpolate above a certain threshold?

DominiqueMakowski, 2022-07-20:
> Should I open a new PR? Sorry I didn't notice these earlier 🙈

No worries, better late than never :) Yes please do open a PR

closed 2022-09-06 by danibene