neuropsychology/NeuroKit #657: Error: the function of hrv_nonlinear(): window cannot contain more data points than the time series

DominiqueMakowski, 2022-06-13:
Hello, 

How can I deal with the error after running following code?

```
analyze_epochs_inter = nk.ecg_intervalrelated(epochs)
```

Thank you in advance!

[image]

DominiqueMakowski, 2022-06-13:
The error seems to originate from `hrv_nonlinear()`. Can you run that function alone and paste the error log? thanks

RyotaTKN, 2022-06-13:
Here is the error by using ```hrv_nonlinear```. (Sorry for small pic)

[image]

DominiqueMakowski, 2022-06-13:
you need to do something like:

```python
for i in epochs:
    peaks = nk.ecg_peaks(epochs[i])
    nk.hrv_nonlinear(peaks)
```

RyotaTKN, 2022-06-13:
Here is the error from the code.

[image]

DominiqueMakowski, 2022-06-13:
How long are your epochs? If there are only a few seconds, I'd suggest using `ecg_eventrelated()` rather than `ecg_intervalrelated()`. The error seems to indicate that the epochs are too short to compute DFA.

That said, I will add a safety-pass so that the function doesn't error

RyotaTKN, 2022-06-13:
The ECG is measured during 100 games, with a time range of 3 to 60 seconds per game.
I would like to somehow extract HRV-related indicators, but is this difficult?

Here is the event table. (sampling_rate = 1000)
[image]

DominiqueMakowski, 2022-06-13:
3 seconds is definitely too short (even 60 seconds) for many of the HRV indices. If you really want to compute something, I would really solely on basic time-domain indicators like RMSSD. Thus I'd recommend only using `hrv_time()`. But there will likelky be a huge effect of the epoch duration.

RyotaTKN, 2022-06-13:
Thank you! I will use hrv_time then.

closed 2022-09-22 by stale[bot]