neuropsychology/NeuroKit #658: ppg_process() fails with high sampling rate

danibene, 2022-06-15:
**Describe the bug**
When the input PPG signal has a high sampling rate, the [ppg_process()](https://neurokit2.readthedocs.io/en/latest/_modules/neurokit2/ppg/ppg_process.html) function does not work:

```python
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
Input In [42], in <cell line: 1>()
----> 1 signals, info = nk.ppg_process(ppg_signal, sampling_rate=sampling_rate)

File ~\repo-dir\lib\NeuroKit\neurokit2\ppg\ppg_process.py:61, in ppg_process(ppg_signal, sampling_rate, **kwargs)
     58 ppg_cleaned = ppg_clean(ppg_signal, sampling_rate=sampling_rate)
     60 # Find peaks
---> 61 info = ppg_findpeaks(ppg_cleaned, sampling_rate=sampling_rate, **kwargs)
     62 info["sampling_rate"] = sampling_rate  # Add sampling rate in dict info
     64 # Mark peaks

File ~\repo-dir\lib\NeuroKit\neurokit2\ppg\ppg_findpeaks.py:65, in ppg_findpeaks(ppg_cleaned, sampling_rate, method, show)
     63 method = method.lower()
     64 if method in ["elgendi"]:
---> 65     peaks = _ppg_findpeaks_elgendi(ppg_cleaned, sampling_rate, show=show)
     66 else:
     67     raise ValueError("`method` not found. Must be one of the following: 'elgendi'.")

File ~\repo-dir\lib\NeuroKit\neurokit2\ppg\ppg_findpeaks.py:116, in _ppg_findpeaks_elgendi(signal, sampling_rate, peakwindow, beatwindow, beatoffset, mindelay, show)
    114 end_waves = np.where(np.logical_and(waves[0:-1], np.logical_not(waves[1:])))[0]
    115 # Throw out wave-ends that precede first wave-start.
--> 116 end_waves = end_waves[end_waves > beg_waves[0]]
    118 # Identify systolic peaks within waves (ignore waves that are too short).
    119 num_waves = min(beg_waves.size, end_waves.size)

IndexError: index 0 is out of bounds for axis 0 with size 0
```

**To Reproduce**
To get the error above, I ran the following code:
```python
sampling_rate = 7500
ppg_signal = nk.ppg.ppg_simulate(sampling_rate=sampling_rate)
signals, info = nk.ppg_process(ppg_signal, sampling_rate=sampling_rate)
```
If I run [signal_filter()](https://neurokit2.readthedocs.io/en/latest/_modules/neurokit2/signal/signal_filter.html) directly...
```python
filtered = nk.signal_filter(
        ppg_signal, sampling_rate=sampling_rate, lowcut=0.5, highcut=8, order=3, method="butter_ba"
    )
```
...I get the following output, containing NaNs:
```python
array([             nan,              nan,              nan, ...,
       -1.02370665e+285, -8.60439443e+284, -7.17490765e+284])
```
**Expected behaviour**
It would be nice for there to be a warning informing me that the sampling rate is too high, or maybe the signal could be resampled to 1000 and then back to the original after filtering?

**System Specifications**

```
- OS: Windows (WindowsPE 64bit) 
- Python: 3.9.12 
- NeuroKit2: 0.2.0 

- NumPy: 1.21.0 
- Pandas: 1.4.2 
- SciPy: 1.7.3 
- sklearn: 1.0.2 
- matplotlib: 3.5.1
```

DominiqueMakowski, 2022-06-16:
But it's strange that `signal_filter` fails in the first place 🤔 and why would it happen just for PPG?

danibene, 2022-06-16:
I think it's the default filtering method in ppg_clean vs. ecg_clean

If I run the same code with `method="butterworth"` instead of `method="butter_ba"`
```python
filtered = nk.signal_filter(
        ppg_signal, sampling_rate=sampling_rate, lowcut=0.5, highcut=8, order=3, method="butterworth"
    )
np.unique(np.isfinite(filtered))
```
There are no NaNs
```python
array([ True])
````

DominiqueMakowski, 2022-06-16:
Good catch! "butterworth_ba" is an older filtering approach then "sos" if I'm not mistaken so we should replace its instances

closed 2022-06-16 by danibene