neuropsychology/NeuroKit #661: ECG peak detection Zong (2003): reimplement

DominiqueMakowski, 2022-06-20:
Our implementation of this peak detection method is tagged as not really working.

https://github.com/neuropsychology/NeuroKit/blob/277f614ee3984647d71ba31aadffea97766aeb0f/neurokit2/ecg/ecg_findpeaks.py#L446-L448

py-ecg-detectors has a [another](https://github.com/berndporr/py-ecg-detectors/blob/36b21ebf21e5344d3d927066a29db84745f0533d/ecgdetectors.py#L553) implementation: we should probably reimplement ours based on that.

Yashaswini278, 2022-07-18:
Hi, could you assign this to me? I'm new to contributing to open source and have some experience in working with ECG data (currently working on an internship in Griffith University to develop an AI-enabled framework for continuous 
monitoring fetal well-being using abdominal ECG signal in order to provide more timely interventions.)

DominiqueMakowski, 2022-07-18:
Hi @Yashaswini278, thanks for volunteering! but actually I think we just recently did it in https://github.com/neuropsychology/NeuroKit/pull/663, but feel free to give it a look and double-check whether everything looks in order (it's currently on the dev branch) :)

Yashaswini278, 2022-07-18:
Oh, sure, thanks! Would love to contribute in other ecg related issues! :)

DominiqueMakowski, 2022-07-18:
Will keep that in mind! Are you interested mostly in ECG signal processing (e.g., peak detection) or more like feature extraction (e.g., HRV etc)?

Yashaswini278, 2022-07-18:
I've mostly worked on peak detection, and using deep learning techniques for abdominal ECG denoising, to obtain fetal ECG. But i'm willing to explore further and work on feature extraction as well!

closed 2022-09-27 by stale[bot]