neuropsychology/NeuroKit #682: Calculate R peaks from RRI

joshgoheen, 2022-07-21:
Hello,

Im looking to analyze HRV using this software with exported data from the EliteHRV app. This app gives heartbeat information in RRI. Ive noticed that the functions for HRV analysis require this information to be in R-peaks. Is there a simple way to convert RRI to R peaks? Ive tried the intervals_to_peaks() function but it seems to give me the r peaks as a running total of the RRI instead of the R-peak information. Im probably missing something simple! Thanks in advance for your help! 

cheers,
Josh

DominiqueMakowski, 2022-07-21:
> but it seems to give me the r peaks as a running total of the RRI instead of the R-peak information

I'm not sure what do you mean? The running sum of RRIs are the R-peak location points

danibene, 2022-09-29:
Now that RRI can be a direct input to the HRV functions (https://github.com/neuropsychology/NeuroKit/pull/710), I'm closing this issue

closed 2022-09-29 by danibene