neuropsychology/NeuroKit #684: RRI & RRI_Time as input to hrv()

danibene, 2022-07-22:
**Describe the solution you'd like**
I would like the HRV functions to be able to accept interbeat intervals and their corresponding timestamps as input rather than the peak samples. This seems to be a recurring issue (e.g. https://github.com/neuropsychology/NeuroKit/issues/682, https://github.com/neuropsychology/NeuroKit/issues/667). I also think it would add more flexibility in terms of how to preprocess the interbeat intervals before HRV analysis (outlier removal & treatment: https://github.com/neuropsychology/NeuroKit/issues/541, interpolating & detrending: https://github.com/neuropsychology/NeuroKit/pull/680).

**How could we do it?**
I did this for a project where I had missing data & therefore the timestamps of the interbeat intervals were needed to properly interpolate in hrv_frequency(). I also made some other changes to the HRV analysis functions to account for the missing data (but I think the input format is probably better to address first on its own). Here is the current state of that branch:
https://github.com/danibene/NeuroKit/blob/feature/adjust_for_missing_intervals/neurokit2/hrv/hrv_time.py#L148-L170

There, I added extra arguments `rri_time` and `data_format` but in retrospect, it would probably be cleaner if there were no extra input arguments and if the input type could just be automatically detected from the keys of a dictionary/columns of a dataframe ("RRI", "RRI_Time")?

Let me know if you are interested in this :)

DominiqueMakowski, 2022-09-21:
Seems I totally missed this (guess the stale bot *can* be useful), what's the status of this?

DominiqueMakowski, 2022-09-21:
Agree to check the peaks argument, best is to keep the number of arguments tight when possible

closed 2022-09-29 by danibene