neuropsychology/NeuroKit #687: KeyError: 'ExpRange'

Ahingsun, 2022-07-31:
**Question and context**

Hi, I'm Inseon. 

I tried to analyze my physio data acquired from BIOPAC MP160. 

I'm now focusing on ECG data and HRV analysis, and Neurokit2 seems easy and reliable analysis tool. Thanks for your work! 

I had the same issue with #593, and I corrected the issue by replacing '_hrv_dfa' as zara6697 suggested (https://github.com/neuropsychology/NeuroKit/issues/593)

After that, my code gave me the following error, which I can't find any issue opened/closed here. 

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.8/dist-packages/neurokit2/bio/bio_analyze.py", line 124, in bio_analyze
    ecg_analyzed = ecg_analyze(ecg_data, sampling_rate=sampling_rate, method=method)
  File "/usr/local/lib/python3.8/dist-packages/neurokit2/ecg/ecg_analyze.py", line 127, in ecg_analyze
    features = ecg_intervalrelated(data, sampling_rate=sampling_rate)
  File "/usr/local/lib/python3.8/dist-packages/neurokit2/ecg/ecg_intervalrelated.py", line 63, in ecg_intervalrelated
    intervals.update(_ecg_intervalrelated_hrv(data, sampling_rate))
  File "/usr/local/lib/python3.8/dist-packages/neurokit2/ecg/ecg_intervalrelated.py", line 128, in _ecg_intervalrelated_hrv
    results = hrv(rpeaks, sampling_rate=sampling_rate)
  File "/usr/local/lib/python3.8/dist-packages/neurokit2/hrv/hrv.py", line 102, in hrv
    out.append(hrv_nonlinear(peaks, sampling_rate=sampling_rate))
  File "/usr/local/lib/python3.8/dist-packages/neurokit2/hrv/hrv_nonlinear.py", line 214, in hrv_nonlinear
    out = _hrv_dfa(peaks, rri, out, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/neurokit2/hrv/hrv_nonlinear.py", line 437, in _hrv_dfa
    out["DFA_alpha1_ExpRange"] = mdfa_alpha1["ExpRange"]
**KeyError: 'ExpRange'**

The error comes when I try

nk.bio_analyze(processed_data, sampling_rate=1000) 
nk.ecg_intervalrelated(ecg_processed, sampling_rate=1000)  
nk.hrv_nonlinear(ecg_processed, sampling_rate=1000) , 

however, the code nk.hrv_time and nk.hrv_frequency correctly provide the results. 

Do you have any idea how to solve this issue? As I'm interested in the non-linear parameters, I really want to get the results. 
I'm attaching my code and data files.

[SMS_fNIRS.txt](https://github.com/neuropsychology/NeuroKit/files/9228717/SMS_fNIRS.txt)
[neurokit2.txt](https://github.com/neuropsychology/NeuroKit/files/9228718/neurokit2.txt)

Thanks in advance. 

Best regards,
Inseon

danibene, 2022-08-01:
Hi @Ahingsun 

Disclaimer: I'm not a maintainer of this package & not sure if I can help here

But I thought I'd try to reproduce your error, and I wasn't able to: https://gist.github.com/danibene/62702dd09eab59f842ede5b8f029e5d3 

Is the example SMS_fNIRS.txt the same data that caused the error?

Ahingsun, 2022-08-02:
> Hi @Ahingsun
> 
> Disclaimer: I'm not a maintainer of this package & not sure if I can help here
> 
> But I thought I'd try to reproduce your error, and I wasn't able to: https://gist.github.com/danibene/62702dd09eab59f842ede5b8f029e5d3
> 
> Is the example SMS_fNIRS.txt the same data that caused the error?

Hi Danibene!

Thanks for trying my code. yes, that's the data that caused the error. 

I just replaced the hrv_nonlinear.py file with the original file (not changing '_hrv_dfa'  part  as suggested in #593), and now it works fine. 

Thanks!

closed 2022-08-02 by Ahingsun

DominiqueMakowski, 2022-08-02:
Thanks @danibene for your help, much appreciated :)