neuropsychology/NeuroKit #700: Attributing the 'Martinez2003' ECG R-peak detection algorithm

peterhcharlton, 2022-09-06:
**Problem**
There is currently no attribution for the 'Martinez2003' ECG R-peak detection algorithm provided in the `_ecg_findpeaks_WT` function, within the [`ecg_findpeaks`](https://github.com/neuropsychology/NeuroKit/blob/master/neurokit2/ecg/ecg_findpeaks.py) function.

**Suggestion**
I think the 'Martinez2003' algorithm is an implementation of the algorithm described in the following paper:
_[Martínez JP, Almeida R, Olmos S, Rocha AP, Laguna P. A wavelet-based ECG delineator: evaluation on standard databases. IEEE Trans Biomed Eng 2004;51:570–581.](https://doi.org/10.1109/TBME.2003.821031)_
This is because the implemented algorithm finds R-peaks using:
- a 1st order Gaussian derivative wavelet transform (see [here](https://pywavelets.readthedocs.io/en/latest/ref/cwt.html?highlight=cwt#gaussian-derivative-wavelets) and Section II.B of the paper)
- scales 2^1 - 2^4 (which is in keeping with those used in Section II.D(1) of the paper)
- change from a negative minimum to a positive maximum (which is similar to the "positive maximum-negative minimum pair" used in Section II.D(1) of the paper).

**Resolution**
- I think mentions of 'Martinez2003' could be replaced with 'Martinez2004' in the following:
   - [`ecg_findpeaks`](https://github.com/neuropsychology/NeuroKit/blob/master/neurokit2/ecg/ecg_findpeaks.py)
   - [`ecg_peaks`](https://github.com/neuropsychology/NeuroKit/blob/master/neurokit2/ecg/ecg_peaks.py)
- I think the following comment in [`ecg_findpeaks`](https://github.com/neuropsychology/NeuroKit/blob/master/neurokit2/ecg/ecg_findpeaks.py) could be updated from 2003 to 2004:
```
# =============================================================================
# Continuous Wavelet Transform (CWT) - Martinez et al. (2003)
# =============================================================================
```
- I think the following documentation provided in [`ecg_peaks`](https://github.com/neuropsychology/NeuroKit/blob/master/neurokit2/ecg/ecg_peaks.py) could be updated from:
```
    * ``martinez2003``
        **Unknown.** Please help us retrieve the correct source!
```
   to:
```
    * ``martinez2004``
        * Martínez, J. P., Almeida, R., Olmos, S., Rocha, A. P., & Laguna, P. (2004) A wavelet-based ECG delineator: evaluation on standard databases. IEEE Trans Biomed Eng, 51(4), 570–581.
        * Gutiérrez-Rivas, R., García, J. J., Marnane, W. P., & Hernández, A. (2015). Novel
          real-time low-complexity QRS complex detector based on adaptive thresholding. IEEE
          Sensors Journal, 15(10), 6036-6043.
```

I'll be happy to raise a pull request for this.

DominiqueMakowski, 2022-09-06:
Good catch! Please don't hestitate to open a PR :)

peterhcharlton, 2022-09-06:
Great thanks - now opened [PR 702](https://github.com/neuropsychology/NeuroKit/pull/702)

peterhcharlton, 2022-09-07:
Thanks for your help with this - I hadn't realised that the tests would need updating too, which @DominiqueMakowski has now done in [PR 702](https://github.com/neuropsychology/NeuroKit/pull/702).

closed 2022-09-07 by peterhcharlton