neuropsychology/NeuroKit #701: Additional PPG peak detectors

peterhcharlton, 2022-09-06:
**The suggestion**
Currently there is one PPG peak detection algorithm available in the [ppg_findpeaks](https://neuropsychology.github.io/NeuroKit/functions/ppg.html#ppg-findpeaks) function, whereas there are several peak detection algorithms available in the [ecg_peaks](https://neuropsychology.github.io/NeuroKit/functions/ecg.html#ecg-peaks) function. A recent [publication](https://iopscience.iop.org/article/10.1088/1361-6579/ac826d) _(disclaimer: written by my colleagues and I)_ found that several PPG peak detection algorithms have been proposed in the literature, and identified the best-performing algorithms (see [here](https://paperswithcode.com/sota/photoplethysmography-ppg-beat-detection-on) for a quick summary). Therefore, I'd suggest implementing additional PPG peak detection algorithms - perhaps including those which have been identified as the best performing.

**How could we do it?**
The recently released [ppg-beats Matlab toolbox](https://ppg-beats.readthedocs.io/) _(disclaimer: written by my colleagues and I)_ contains Matlab implementations of several PPG peak detection algorithms. We could translate the relevant algorithms from this Matlab code into Python code for inclusion in the [ppg_findpeaks](https://neuropsychology.github.io/NeuroKit/functions/ppg.html#ppg-findpeaks) function. I'd be happy to work on this if helpful.

DominiqueMakowski, 2022-09-06:
I think that'd be super helpful, actually, as NK's PPG-related functions are a bit under-developped (as compared to ECG). + It will be a good way to let NK users discover your work. Feel free to open a PR and don't hesitate to credit yourself & your team; and your own matlab software where it's due ;) (for example, in the references, the description of the function, as comments in the code etc.)

peterhcharlton, 2022-09-07:
Great, thanks, will do. This will take me a bit of time. I'll aim to contribute a PPG beat detector in the next few weeks.

LegrandNico, 2022-09-15:
I was planning to translate the Multi-Scale Peak and Trough Detection algorithm for [Systole](https://github.com/embodied-computation-group/systole) in the next weeks from your Matlab code @peterhcharlton , so I am happy to help with this one for NeuroKit 👍

peterhcharlton, 2022-09-15:
Excellent, many thanks. In my opinion there are two key PPG beat detectors which ideally we would implement - this one (which I tend to abbreviate as MSPTD), and 'qppg'. That's great that you'll aim to work on MSPTD. Thanks again

closed 2023-02-18 by stale[bot]