neuropsychology/NeuroKit #714: Choice of dimension parameter for entropy measures

JoseAlanis, 2022-09-28:
I noticed that the defaults for the dimension parameter are not consistent across functions to compute entropy measures and was wondering whether this was a deliberate choice.

For instance:
- [entropy_permutation()](https://neuropsychology.github.io/NeuroKit/functions/complexity.html#entropy-permutation) uses a default dimension parameter of 3.
- [entropy_multiscale()](https://neuropsychology.github.io/NeuroKit/functions/complexity.html#entropy-multiscale) uses a default dimension parameter of 2.

Sorry if this a silly question, I'm just starting to use the package (which I think is great!) and trying to better understand the methods.

DominiqueMakowski, 2022-09-28:
mmh no that's not on purpose, actually the default for multiscale could probably be 3 too. Feel free to make a PR to fix that :)

JoseAlanis, 2022-09-28:
Ok, thanks for the reply, I'll take care of it :)

JoseAlanis, 2022-10-03:
took care of this #715 😄 . I guess this issue can be closed then?

closed 2022-10-03 by DominiqueMakowski