neuropsychology/NeuroKit #722: How to work with matlab data

aXcELLcore, 2022-10-03:
**Question and context**
Hi, I try to import my data analysis from matlab to neurokit but once its loaded I have not found any function in NeuroKit to be able to work with them, I would like that you can help me to solve this problem

danibene, 2022-10-04:
Hi @aXcELLcore have you tried this scipy function? https://docs.scipy.org/doc/scipy/reference/generated/scipy.io.loadmat.html

aXcELLcore, 2022-10-04:
Thank you very much for the help, it has been very useful to me to be able to access my data, if it is not too much trouble, could you guide me now how I can incorporate these data or use them in NeuroKit

danibene, 2022-10-04:
Could you please provide more information on what you're trying to do and the exact step where you're stuck? It's difficult for me to judge whether I would easily be able to answer your question without more context.

aXcELLcore, 2022-10-05:
I would like to be able to obtain the global field power, the power and the power spectral density, thank you very much in advance for your advice

danibene, 2022-10-05:
Have you consulted the documentation? E.g. https://neuropsychology.github.io/NeuroKit/functions/eeg.html

closed 2022-10-11 by DominiqueMakowski