neuropsychology/NeuroKit #729: "[N] not in index" exception raised by ecg_delineate

citron, 2022-10-11:
**Bug Description:**
ECG having length N, delineation stops on an exception with "[N] not in index" as a debug message.

**To Reproduce:**

1. save following code as debug_pyatsa.py in your favorite folder
```
import neurokit2 as nk
import pandas as pd
import numpy as np
import os

HERTZ = 250
filin = "debug.csv"
ecg_signal = pd.read_csv(filin, header=0, names=["ECG"])['ECG']
ecg_cleaned = nk.ecg_clean(ecg_signal, sampling_rate=HERTZ)
_, rpeaks = nk.ecg_peaks(ecg_cleaned, sampling_rate=HERTZ)
_, waves_peak = nk.ecg_delineate(ecg_cleaned, rpeaks, sampling_rate=HERTZ, method="peak", show=False, show_type='peaks')
```
2. save attached ECG sample as debug.csv 
[debug.csv](https://github.com/neuropsychology/NeuroKit/files/9758396/debug.csv)
3. python debug_pyatsa.py
4. See error

**Expected behaviour:**
No exception should occur as the CSV input file is OK.

**Screenshots**
[image]
[image]

**System Specifications:**

- OS: Linux (ELF 64bit) 
- Python: 3.10.7 
- NeuroKit2: 0.2.1 
- NumPy: 1.23.3 
- Pandas: 1.5.0 
- SciPy: 1.9.1 
- sklearn: 1.1.2 
- matplotlib: 3.6.1

same problem on Windows :
 - OS: Windows (WindowsPE 64bit) 
- Python: 3.10.7 
- NeuroKit2: 0.2.1 
- NumPy: 1.23.3 
- Pandas: 1.5.0 
- SciPy: 1.9.2 
- sklearn: 1.1.2 
- matplotlib: 3.6.1

DominiqueMakowski, 2022-10-12:
@Tam-Pham time to revise your python skills 😁

closed 2022-10-14 by danibene

citron, 2022-10-14:
I do confirm. It works now using the dev branch. Thank you very much.