neuropsychology/NeuroKit #733: Distance between peaks

mn5k, 2022-10-16:
Thanks for your great work.

I have a question about [_signal_findpeaks_distances](https://github.com/neuropsychology/NeuroKit/blob/277f614ee3984647d71ba31aadffea97766aeb0f/neurokit2/signal/signal_findpeaks.py#L188).
- I think this function calculates two distances (distance between i-th and (i-1)-th elements and i-th and (i+1)-th elements) and selects the smaller one. Is this right?
- In this case, we should reverse the order of distances_prev.

```
def _signal_findpeaks_distances(peaks):

    if len(peaks) <= 2:
        distances = np.full(len(peaks), np.nan)
    else:
        distances_next = np.concatenate([[np.nan], np.abs(np.diff(peaks))])
        distances_prev = np.concatenate([np.abs(np.diff(peaks[::-1])), [np.nan]])
        distances = np.array([np.nanmin(i) for i in list(zip(distances_next, distances_prev))])

    return distances

def _signal_findpeaks_distances_2(peaks):

    if len(peaks) <= 2:
        distances = np.full(len(peaks), np.nan)
    else:
        distances_next = np.concatenate([[np.nan], np.abs(np.diff(peaks))])
        distances_prev = np.concatenate([np.abs(np.diff(peaks[::-1]))[::-1], [np.nan]])
        distances = np.array([np.nanmin(i) for i in list(zip(distances_next, distances_prev))])
        # or distances = np.insert(distances_next, 0, distances_next[0])

    return distances

peaks = np.array([1, 10, 10**2, 10**3, 10**4], dtype=np.float32)
print(_signal_findpeaks_distances(peaks))  # [9000.    9.   90.    9. 9000.]
print(_signal_findpeaks_distances_2(peaks))  # [   9.    9.   90.  900. 9000.]

peaks = np.array([1, 2, 4, 7, 0], dtype=np.float32)
print(_signal_findpeaks_distances(peaks))  # [7. 1. 2. 1. 7.]
print(_signal_findpeaks_distances_2(peaks))  # [1. 1. 2. 3. 7.]
```

danibene, 2022-10-19:
@mn5k thank you for catching this!

> * I think this function calculates two distances (distance between i-th and (i-1)-th elements and i-th and (i+1)-th elements) and selects the smaller one. Is this right?

Yes, I understood it this way too.

> * In this case, we should reverse the order of distances_prev.

This change makes sense to me. Would you be willing to open a pull request? :)

danibene, 2022-10-19:
We could alternatively use the already computed distances from `distance_next` to save some computation:

```

def _signal_findpeaks_distances(peaks):

    if len(peaks) <= 2:
        distances = np.full(len(peaks), np.nan)
    else:
        distances_next = np.concatenate([[np.nan], np.abs(np.diff(peaks))])
        distances_prev = np.concatenate([np.abs(np.diff(peaks[::-1])), [np.nan]])
        distances = np.array([np.nanmin(i) for i in list(zip(distances_next, distances_prev))])

    return distances

def _signal_findpeaks_distances_2(peaks):

    if len(peaks) <= 2:
        distances = np.full(len(peaks), np.nan)
    else:
        distances_next = np.concatenate([[np.nan], np.abs(np.diff(peaks))])
        distances_prev = np.concatenate([np.abs(np.diff(peaks[::-1]))[::-1], [np.nan]])
        distances = np.array([np.nanmin(i) for i in list(zip(distances_next, distances_prev))])
        # or distances = np.insert(distances_next, 0, distances_next[0])

    return distances

def _signal_findpeaks_distances_3(peaks):

    if len(peaks) <= 2:
        distances = np.full(len(peaks), np.nan)
    else:
        distances_next = np.concatenate([[np.nan], np.abs(np.diff(peaks))])
        distances_prev = np.concatenate([distances_next[1:], [np.nan]])
        distances = np.array([np.nanmin(i) for i in list(zip(distances_next, distances_prev))])
        # or distances = np.insert(distances_next, 0, distances_next[0])

    return distances

peaks = np.array([1, 10, 10**2, 10**3, 10**4], dtype=np.float32)
print(_signal_findpeaks_distances(peaks))  # [9000.    9.   90.    9. 9000.]
print(_signal_findpeaks_distances_2(peaks))  # [   9.    9.   90.  900. 9000.]
print(_signal_findpeaks_distances_3(peaks))  # [   9.    9.   90.  900. 9000.]

peaks = np.array([1, 2, 4, 7, 0], dtype=np.float32)
print(_signal_findpeaks_distances(peaks))  # [7. 1. 2. 1. 7.]
print(_signal_findpeaks_distances_2(peaks))  # [1. 1. 2. 3. 7.]
print(_signal_findpeaks_distances_3(peaks))  # [1. 1. 2. 3. 7.]

```

mn5k, 2022-10-20:
@danibene Sure. I'll create a PR.

PR #738 (2022-10-31): [fix] fix _signal_findpeaks_distances (#733)
# Description

- fix _signal_findpeaks_distances

# Proposed Changes
- Closes #733 
- I changed the `_signal_findpeaks_distances` calculates two distances (distance between i-th and (i-1)-th elements and i-th and (i+1)-th elements) and selects the smaller one.

# Checklist

Here are some things to check before creating the PR. If you encounter any issues, do let us know :)

- [x] I have read the [CONTRIBUTING](https://github.com/neuropsychology/NeuroKit/blob/master/.github/CONTRIBUTING.rst#structure-and-code) file.
- [x] My PR is targetted at the ** [... cut]
closed 2022-10-31 by danibene