neuropsychology/NeuroKit #751: ecg_peaks() method=""

nileshkrsahu, 2022-11-16:
**Question and context**
In the reference given on the neuorkit website, the paper by Gutiérrez-Rivas("Gutiérrez-Rivas, R., García, J. J., Marnane, W. P., & Hernández, A. (2015). Novel real-time low-complexity QRS complex detector based on adaptive thresholding. IEEE Sensors Journal, 15(10), 6036-6043." ) was cited, but I didn't find any method for this.

DominiqueMakowski, 2022-11-16:
Hi @nileshkrsahu, could you help us link to where we mentioned that paper?

nileshkrsahu, 2022-11-16:
Here is the link: https://neuropsychology.github.io/NeuroKit/functions/ecg.html#ecg-peaks
In references of ecg_peaks (Reference 14)

DominiqueMakowski, 2022-11-16:
@levsilva if I'm not mistaken you recently sanitized a bit all the ECG-peak detection references and all that, would you know something here?

levsilva, 2022-11-16:
Hi @DominiqueMakowski and @nileshkrsahu,

@nileshkrsahu In the file `ecg_findpeaks.py`, you will find comments within each hidden function of individual peak finders, referring to the papers related to them. The function for the method `rodrigues2021` has a comment saying it was inspired by on _Gutierrez-Rivas (2015)_ and _Sadhukhan (2012)_ papers. So those two papers are related to this method.

@DominiqueMakowski In fact, I have updated the doc for ecg_peaks.py, pooling and itemizing all paper related to each method. Maybe some have updated it after me?

closed 2023-02-05 by stale[bot]