neuropsychology/NeuroKit #759: EOG_Rate_Mean

LeilaNS, 2022-12-15:
**Question and context**
Hi,
I used nk2 to calculate the number of blinks and the blink rate from my 4 minute EOG signals. Here is my code:
```
processed_eog,eog_info=nk.eog_process(eog_signal, sampling_rate=Fs)
results = nk.bio_analyze(processed_eog, sampling_rate=Fs)

```
I checked the blink detection, it works fine, however I can't figure out what the definition of blink rate is? Why isn't the minimum blink rate simply the number of blinks / total duration? Is this a derivative of Blinks?
```

results['EOG_Peaks_N'].values
Out[240]: array([17.])
results['EOG_Rate_Mean'].values
Out[241]: array([7.20688531])
len(processed_eog)/Fs
Out[247]: 240.01
```
Best

danibene, 2022-12-17:
Currently the blink rate is computed with the mean of the interpolated instantaneous blink rate computed with the interval between each pair of subsequent blinks using [`signal_rate()`](https://github.com/neuropsychology/NeuroKit/blob/a1f317bf620b1d3534b1a7c37a1e05f0044b6837/neurokit2/signal/signal_rate.py):
https://github.com/neuropsychology/NeuroKit/blob/2fac6ebd7427d71b70fda7501893d1a0c478a8a7/neurokit2/eog/eog_process.py#L78

@DominiqueMakowski I was wondering if the decision to define blink rate this way was intentional for EOG data or just for consistency with the code for the rest of the signals? What do you think of redefining mean blink rate as the number of blinks / total duration?

See [this paper](https://www.ncbi.nlm.nih.gov/pmc/articles/PMC8068990/):

> We define blink rate as the number of blinks per minute. Our blink detection algorithm always assigns a single maximum frame to each blink. We compute the average blink rate for the dataset as the number of blink maxima divided by the total length of the dataset in minutes. Users can easily detect instantaneous blink rates by computing the number of blinks in sliding window. Literature suggests that instantaneous blink rates be averaged over intervals of at least 3 to 5 min ([Zaman and Doughty, 1997](https://www.frontiersin.org/articles/10.3389/fnins.2017.00012/full#B47)).

Also [this paper](https://www.frontiersin.org/articles/10.3389/fpsyg.2022.788231/full):

> In Experiment 1 and 2, sEBR was computed by dividing the total number of blinks by the total period duration for any given phase resulting in units of blinks per minute

DominiqueMakowski, 2022-12-18:
I think it's implemented like this similar to what happens in other signals (e.g., the mean heart rate is also the mean of the interpolated rate signal). 

Now whether that makes sense as compared to a more straightforward `number of events / minute`, yes I think there are arguments for that. In short, by computing the continuous rate, we are trying to approximate the underlying continuous activity that manifests as discrete events (blinks, beats, etc.). The fundamental assumption is that behind these momentary events, there is some dynamic system that has a continuous activity (for instance related to the nervous system). 

Note that this assumption is not necessarily always totally verified (typically, blinks can be triggered by external events - e.g., a blowing of air, in which case it does not anymore relate to some underlying process - e.g., the attention level). 

But for what it's worth, in the majority of typical scenarios and controlled environments, the underlying continuous activity is what is of interest. Importantly, it allows us to correct things (e.g., filter the rate, interpolate missing events, etc.) to remove potential artefacts. In the end, taking the average of the continuous activity makes (usually) sense theoretically + can give a "more true" estimate than `n_blinks / minute` which can be contaminated by artefacts.
 
That being said, I'm sure there are plenty cases in which people would be interested to compute the average raw blink rate (which should be - in most cases - very very close to the mean of the interpolated), and it should be pretty straightforward to do explicitly

@danibene what do you think?

closed 2023-03-18 by stale[bot]