neuropsychology/NeuroKit #770: How to combine RRi and R-peak variation of ECG for a better EDR signal?

mobilemodem, 2023-02-01:
I am trying to improve similarity of EDR signal to respiration signal.
I used two traditional method to get EDR signal:
1\ R-peak amplitude variation to get AM signal of thor respiration. Call EDR1 later.
2\ RRi to get RSA signal. Call EDR2 later.
I calculate the correlation between EDR and RESP signal on Fantasia database.
Some show good similarity on EDR1, Some on EDR2.
The best corrrelation value of EDR1 is 0.925, the worst is 0.041, averagely 0.67 on 40 files.
The best corrrelation value of EDR2 is 0.757, the worst is 0.171, averagely 0.451 on 40 files.
For file f2y05, r(EDR2|rsp) = 0.171, r(EDR1|rsp) = 0.741.
For file f2o02, r(EDR2|rsp) = 0.655, r(EDR1|rsp) = 0.041.

How can I decide the best EDR signal without resp signal for reference?

DominiqueMakowski, 2023-02-01:
First, note that RSA is undergoing some improvements (@peterhcharlton), which might partly explain why the correlation is lower. Moreover, it seems based on your results that EDR1 is on average better than the one based on the old RSA? So I'd go with that

mobilemodem, 2023-04-26:
> 

I am trying some ideas to understand better the differences between diffrent EDR signals. But no progress yet.
Hope this issue kept open for future discussion.

closed 2023-08-07 by stale[bot]