neuropsychology/NeuroKit #772: Strange indexing error using "ecg_process"

niwolpert, 2023-02-02:
**Bug description**
I'm calling 'ecg_process' on an epochs of ECG signals of 5000 samples each (sample frequency = 250 Hz). With most epochs this works fine, but with some epochs, it results in the error "KeyError: '[5000] not in index'" (here 5000 corresponds to the number of samples, if I change the epoch length, it throws the same error). I cannot explain why it throws this error with some epochs. There are no Nan in the epochs and the ECG signal looks clean and regular.

[image]

**To Reproduce**

import pandas as pd
ecg_signal = pd.read_csv('ecg_signal.csv')        # load csv provided in attachment
df_raw, info = nk.ecg_process(ecg_signal, sampling_rate=250)

**Expected behaviour**
I expect the function to return me the raw df and info dictionary.

**System Specifications**

- OS: Windows (WindowsPE 64bit) 
- Python: 3.9.12 
- NeuroKit2: 0.2.0 
- NumPy: 1.22.3 
- Pandas: 1.4.4 
- SciPy: 1.7.3 
- sklearn: 1.0.2 
- matplotlib: 3.5.2

[ecg_signal.csv](https://github.com/neuropsychology/NeuroKit/files/10567836/ecg_signal.csv)

danibene, 2023-02-13:
Hi, thanks for pointing this out!

I think this is happening because `ecg_delineate()` detects points in the segmented ECG, and since the last heartbeat segment has zeros appended at the end (I think that's done so that the length of that last segment isn't shorter than the previous segments), it is possible that the point indices detected don't actually correspond to indices in the original ECG signal. One possible way to deal with this would be to remove the appended zeros from the final heartbeat:
https://github.com/danibene/NeuroKit/commit/1bedb50dc7dc65f51533fcc0b6cefecc5da1a722

Or another way could be to go through all the points detected in `ecg_delineate()` and remove any indices that are higher than the last index in the original signal.

Any thoughts @DominiqueMakowski ?

DominiqueMakowski, 2023-02-13:
> Or another way could be to go through all the points detected in ecg_delineate() and remove any indices that are higher than the last index in the original signal.

Yes, I think we could add that by default in ecg_delinate() as a sanity postprocess where it removes all invalid indices

DominiqueMakowski, 2023-02-17:
Should be fixed on the dev version thanks to @danibene :)

closed 2023-02-17 by DominiqueMakowski

niwolpert, 2023-02-18:
Awesome, thanks a lot!