neuropsychology/NeuroKit #773: RR peaks extraction on Shaoxing ECG dataset leads to an error

srulikbd, 2023-02-08:
**Describe the bug**

I'm working on this dataset:
[https://physionet.org/content/ecg-arrhythmia/1.0.0/](url)

The data is already cleaned and comes in 500Hz sampling rate.

when try the standard method for RR extraction:

`signals, info = nk.ecg_process(data, sampling_rate=sample_rate)`

It leads to the following error:

`C:\Users\User\PycharmProjects\pythonProject\walkky-ML\lib\site-packages\neurokit2\signal\signal_period.py:68: NeuroKitWarning: Too few peaks detected to compute the rate. Returning empty vector.
  warn(
Traceback (most recent call last):
  File "C:\Users\User\Desktop\Work\Walkky\Code\walkky-ML\classifier\ml\processs_data\create_rr_index.py", line 73, in <module>
    create_heartbeat_from_shaoxing_dataset_to_csv(lead_index)
  File "C:\Users\User\Desktop\Work\Walkky\Code\walkky-ML\classifier\ml\processs_data\create_rr_index.py", line 50, in create_heartbeat_from_shaoxing_dataset_to_csv
    signals, info = extract_heart_beats(df[lead].to_numpy(), sample_rate)
  File "C:\Users\User\Desktop\Work\Walkky\Code\walkky-ML\classifier\ml\processs_data\load_physionet.py", line 33, in extract_heart_beats
    signals, info = nk.ecg_process(data, sampling_rate=sample_rate)
  File "C:\Users\User\PycharmProjects\pythonProject\walkky-ML\lib\site-packages\neurokit2\ecg\ecg_process.py", line 98, in ecg_process
    quality = ecg_quality(ecg_cleaned, rpeaks=rpeaks["ECG_R_Peaks"], sampling_rate=sampling_rate)
  File "C:\Users\User\PycharmProjects\pythonProject\walkky-ML\lib\site-packages\neurokit2\ecg\ecg_quality.py", line 103, in ecg_quality
    quality = _ecg_quality_averageQRS(ecg_cleaned, rpeaks=rpeaks, sampling_rate=sampling_rate)
  File "C:\Users\User\PycharmProjects\pythonProject\walkky-ML\lib\site-packages\neurokit2\ecg\ecg_quality.py", line 133, in _ecg_quality_averageQRS
    heartbeats = ecg_segment(ecg_cleaned, rpeaks, sampling_rate)
  File "C:\Users\User\PycharmProjects\pythonProject\walkky-ML\lib\site-packages\neurokit2\ecg\ecg_segment.py", line 54, in ecg_segment
    heartbeats = epochs_create(
  File "C:\Users\User\PycharmProjects\pythonProject\walkky-ML\lib\site-packages\neurokit2\epochs\epochs_create.py", line 157, in epochs_create
    epoch_max_duration = int(max((i * sampling_rate for i in parameters["duration"])))
ValueError: cannot convert float NaN to integer`

I tried normalize the data but it didn't help.

**To Reproduce**
download sample of Shaoxing dataset and try to do RR extraction.

**Expected behaviour**
extract the RR index.

**System Specifications**

`- OS: Windows (WindowsPE 64bit) 
- Python: 3.9.5 
- NeuroKit2: 0.2.2 

- NumPy: 1.22.4 
- Pandas: 1.2.4 
- SciPy: 1.6.3 
- sklearn: 1.2.0 
- matplotlib: 3.4.2
`

DominiqueMakowski, 2023-02-08:
Hi, could you attach here directly a portion of data that causes the problem (as csv or txt) so we can more easily debug? Thanks!

srulikbd, 2023-02-08:
[shaoxing_sample.csv](https://github.com/neuropsychology/NeuroKit/files/10687642/shaoxing_sample.csv)
Thanks for your wonderful library!!

srulikbd, 2023-02-09:
It seems that with ecg_peaks() method it find only 2 peaks.

Thanks for your fast answers!!

closed 2023-02-09 by DominiqueMakowski