neuropsychology/NeuroKit #774: Question about identify the individual heartbeat in different leads

Lospub, 2023-02-08:
I am trying to segment an ECG signal into single heartbeats using ecg_segment method.
My question is when doing the segment for different ECG leads, is there any different argument needed to use this method? Because it looks like the segment method sometimes can only perform some leads well. 
Or is there any other method I can use to segment an ECG signal into single heartbeats?

DominiqueMakowski, 2023-02-08:
Hi! The most "clean" way to segment would be to delineate and get the onset of the P wave and the end of the T wave, but then you might end up with segments of different size. Plus delineating is not straightforward and often fails. So a more robust method is to make sure you get the R-peak located right and then cut a window around... but indeed this is problematic as the length of a beat can depend on a variety of things. 

All in all I am afraid we don't have a full proof method extracting individual heart beats

Lospub, 2023-02-13:
I see, thanks for your clarification.

closed 2023-02-13 by Lospub