neuropsychology/NeuroKit #778: eda_sympathetic() assumes the sampling frequency of 400Hz

GanshengT, 2023-02-14:
**Describe the bug**
I found the eda_sympathetic() function assumes the sampling frequency is 400Hz when calculating sympathetic index from EDA signal. In the original paper, the authors say "Before down-sampling, the data was filtered with an 8th-order Chebyshev Type I low-pass filter (0.8 Hz). Down-sampling from 400 to 2 Hz was performed in two steps (using consecutive down-sampling factors of 1/20 and 1/10, respectively). Finally, signals were high-pass filtered (0.01 Hz, Butterworth, 8th order) to remove any trend."

eda_sympathetic() calls _eda_sympathetic_posada to calculate the sympathetic index:
elif method.lower() in ["posada", "posada-quintero", "quintero"]:
        out = _eda_sympathetic_posada(eda_signal, frequency_band=frequency_band, show=show)
    
When looking at _eda_sympathetic_posada, you will find signal was downsampled twice (200folds). The problem is that the signal_filter assumes the sampling rate of the downsampled signal is 2.

def _eda_sympathetic_posada(eda_signal, frequency_band=[0.045, 0.25], show=True, out={}):

    # First step of downsampling
    downsampled_1 = scipy.signal.decimate(eda_signal, q=10, n=8)  # Keep every 10th sample
    downsampled_2 = scipy.signal.decimate(downsampled_1, q=20, n=8)  # Keep every 20th sample

    # High pass filter
    eda_filtered = signal_filter(
        downsampled_2, sampling_rate=2, lowcut=0.01, highcut=None, method="butterworth", order=8
    )

Reference:
Posada-Quintero, H. F., Florian, J. P., Orjuela-Cañón, A. D., Aljama-Corrales, T.,
      Charleston-Villalobos, S., & Chon, K. H. (2016). Power spectral density analysis of
      electrodermal activity for sympathetic function assessment. Annals of biomedical engineering,
      44(10), 3124-3135.

**To Reproduce**
eda = nk.data('bio_resting_8min_100hz')['EDA']
indexes_posada = nk.eda_sympathetic(eda, sampling_rate=100, method='posada', show=True)

**Expected behaviour**
The eda signal is sampled at 100Hz. The high pass filter would need the input sampling_rate = sampling_rate/200, which is 0.5

**Screenshots**
If applicable and useful, you can add screenshots by drag-and-dropping the images in the issue window.

**System Specifications**

It's important that you give us some information about the system you are using. For that you can run:

```python
import neurokit2 as nk
nk.version()
```

- OS: Darwin ( 64bit) 
- Python: 3.10.8 
- NeuroKit2: 0.2.3 

- NumPy: 1.24.2 
- Pandas: 1.5.2 
- SciPy: 1.10.0 
- sklearn: 1.2.1 
- matplotlib: 3.5.3

GanshengT, 2023-02-14:
Another concern is about the title of the spetral plot of the EDA:
Please go to https://neuropsychology.github.io/NeuroKit/_modules/neurokit2/eda/eda_sympathetic.html#eda_sympathetic and check the source code
 if show is True:
        ax = psd_plot.plot(x="Frequency", y="Power", title="EDA Power Spectral Density (ms^2/Hz)")
        ax.set(xlabel="Frequency (Hz)", ylabel="Spectrum")

The unit of skin conductance is generally uS, which I believe, is the case for eda = nk.data('bio_resting_8min_100hz')['EDA'].

For unnormalized psd, the title should be uS(Simons)/Hz.

DominiqueMakowski, 2023-02-15:
Hi @GanshengT thanks a lot for all these input. I think they are all easily fixable indeed, would you be interested in contributing a PR to fix them all :) ?

GanshengT, 2023-02-16:
I am happy to. I was thinking that we downsample the input signal to 2Hz and remind the user that the sampling frequency in the original paper is 400Hz. Let me know if that's reasonable.

GanshengT, 2023-02-17:
Changes have been committed to solve the issue.

closed 2023-02-17 by GanshengT