neuropsychology/NeuroKit #782: cvxEDA seems to fail to decompose phasic and tonic component of the skin conductance signal

GanshengT, 2023-02-16:
This serves as an extended discussion about cvxEDA #781 
I have no trouble runing cvxEDA. But the results are not expected. The decomposed tonic and phasic components are different from other methods.
To reproduce what I saw kindly import the following package
import neurokit2 as nk
import scipy.io
import numpy as np
import seaborn as sns
import pandas as pd
import ledapy

eda_signal = nk.data("bio_eventrelated_100hz")["EDA"]
# Preprocess
eda_cleaned = nk.eda_clean(eda_signal, sampling_rate=sfreq, method='neurokit') # low pass filter
# median convolution
eda_decomposed = nk.eda_phasic(eda_cleaned, sampling_rate=sfreq, method = 'median')
data2plot = eda_decomposed
data2plot['raw'] = eda_cleaned
# filter
eda_decomposed = nk.eda_phasic(eda_cleaned, sampling_rate=sfreq, method = 'highpass')
data2plot['highpass_tonic'] = eda_decomposed['EDA_Tonic']
data2plot['highpass_phasic'] = eda_decomposed['EDA_Phasic']
# cvx (results look bad)
eda_decomposed = nk.eda_phasic(eda_cleaned, sampling_rate=sfreq, method = 'cvxeda')
data2plot['cvx_tonic'] = eda_decomposed['EDA_Tonic']
data2plot['cvx_phasic'] = eda_decomposed['EDA_Phasic']
# ledalab, continuous decomposition
phasicdata = ledapy.runner.getResult(eda_cleaned, 'phasicdata', sfreq/4, downsample=1, optimisation=2)
data2plot['cda_phasic'] = phasicdata
data2plot['cda_tonic'] = eda_cleaned - phasicdata
fig = nk.signal_plot(data2plot)
# plt.plot(eda_decomposed)
[image]

fig = nk.signal_plot(data2plot[['raw','EDA_Tonic', 'highpass_tonic', 'cda_tonic', 'cvx_tonic']])
[image]
[image]

- OS: Darwin ( 64bit) 
- Python: 3.10.8 
- NeuroKit2: 0.2.3 

- NumPy: 1.24.2 
- Pandas: 1.5.2 
- SciPy: 1.10.0 
- sklearn: 1.2.1 
- matplotlib: 3.5.3

lciti, 2023-02-16:
Actually, using your code, I get decent results (of course different algorithms may have a different cutoff about what they consider tonic and what phasic). See below.
[image]

I have used [test.py.zip](https://github.com/neuropsychology/NeuroKit/files/10759715/test.py.zip) to generate the plots.

I think I was too fast to dismiss this error as significantly different from https://github.com/neuropsychology/NeuroKit/issues/781 . Since in both cases the same code and data works for me but not for others, it could be an issue with versions of the cvxopt code or its dependencies. @GanshengT ,is there any chance you could try on a different platform (either windows or linux)? Could you please use this code [versions.py.zip](https://github.com/neuropsychology/NeuroKit/files/10759711/versions.py.zip) so I can try to identify the exact build that you are using?

GanshengT, 2023-02-16:
Thanks for the check
Here is the platform information:
platform macOS-10.16-x86_64-i386-64bit
python 3.10.8 (main, Nov 24 2022, 08:09:04) [Clang 14.0.6 ]
neurokit2 0.2.3
cvxopt 1.3.0
8f3d628339aee2c0f80a6131b1b976f78fe93dddbfc95ed0534abe688c6cc518 gsl.cpython-310-darwin.so
52a38b90d2133e515f5984a2ac6349a1f76cc2c4419f3c340abe7bae569a012e misc.py
6601c3e9827d30b9bb259d561dc1c1372c829743598052ad5992ef7e50192875 misc_solvers.cpython-310-darwin.so
ce7390223aa7a26311677e1e41d740233a6679218885be2fbb1ae8462af00199 dsdp.cpython-310-darwin.so
6f886e06a94cdec4396614a1a1fd47c0b05ab675f7c40244bdf81c9229ee2be7 _version.py
ef6ef92fe839d42f39df802dc58a6161cc8d616bd77f2df4be224cc881ef0429 amd.cpython-310-darwin.so
af9c67739f0d0be9d57bdd04e31bfa749d5b62eb8d6457cbe5359211144aced8 base.cpython-310-darwin.so
c991b8009bd51d6a94536c185e772f5f91c44106c2bed8beee922d3f35a07e91 cholmod.cpython-310-darwin.so
54d84e0738cfcde2117bf8e28e85980ddebbc0dab9f44285143cef3b1efc06c4 msk.py
05ff52326929493c8c04e502eb77369778f840fba4c1eebdaf87285c2ce914bc __init__.py
5293e05946cef55a31a4ae85eaafff2e624e7f9202fd437523280152cfe64bf6 solvers.py
88529236210c868b2d77f7c86f6b15222e974890dbdcf1afeab65919e9707478 lapack.cpython-310-darwin.so
f71ff035ab5bb322263ce6809b6c4d18d73a055970c9c52706e970f22fd4f954 glpk.cpython-310-darwin.so
7f6f14169c90ad48c3b73efbce0f1fb2dfc4921ebbe157774ebdd2cf8996b955 blas.cpython-310-darwin.so
39f11539941ca86642a08ee01649ef5e0ecab7c2c74b5fda2b349e40e75f0f3b fftw.cpython-310-darwin.so
046d7cd38ac933230fdec08dcc6f7cfb7713ea769f7f736bc4683597c99dc8df umfpack.cpython-310-darwin.so
693914096952d09b637df5c7d7f7544e5651dc09253338871af6583009ff0bb9 cvxprog.py
b5af280e6ca46399ed67094e43cb89a3b462f5e3f12e6487a44a756d20c8ab5b modeling.py
025042af47b2fbeab2ab638373167780a5147566a4ccf67f6cbc3cae65ae3fd3 info.py
f326385725e4c6a324d3c9677107eed7173b8b22563dfd0eb580aff3bda8110f coneprog.py
aa588c9feab2a0f782da01777ff71d110070b89ce52bc4b600fcaab8689ec1b0 printing.py

closed 2023-02-16 by GanshengT

lciti, 2023-02-16:
@GanshengT I am glad that it works now. Can you help debug what could have gone wrong? From the checksums that you uploaded before (8f3d62... gsl.cpython-310-darwin.so), it looks like you were already using the most up-to-date version [cvxopt-1.3.0-py310h7bc8341_4.tar.bz2](https://anaconda.org/conda-forge/cvxopt/1.3.0/download/osx-64/cvxopt-1.3.0-py310h7bc8341_4.tar.bz2). So maybe when you updated using `conda` or `pip` something else was upgraded. Do you happen to know (from the text produced by the update command) which packages were updated (and from which version to which version)? Thanks! (No problem if you have already closed that shell and lost that info).

GanshengT, 2023-02-17:
Sorry I have lost the information. I will pay attention to the system information if the error occurs again.