neuropsychology/NeuroKit #784: nk.eda_simulate non reproducible even when random_state is set to a constant

lciti, 2023-02-17:
**Describe the bug**
`nk.eda_simulate` generates random signals even when `random_state` is set to a constant

**To Reproduce**
```python
>>> eda_signal = nk.eda_simulate(duration=30, scr_number=5, drift=0.1, random_state=1234)
>>> print(eda_signal)
[0.98571118 0.98526776 0.98387649 ... 3.9812714  3.9785683  3.97764784]
>>> eda_signal = nk.eda_simulate(duration=30, scr_number=5, drift=0.1, random_state=1234)
>>> print(eda_signal)
[0.99979647 0.99998727 1.00034615 ... 4.05665343 4.05670009 4.05676347]
```

**Expected behaviour**
Setting `random_state` should make `nk.eda_simulate`  always return the same signal.
The problem is that `nk.eda_simulate` calls `signal_distort` (line [89](https://github.com/neuropsychology/NeuroKit/blob/8c564938c5fc2b4a377fd89bb14ac07a430983b9/neurokit2/eda/eda_simulate.py#L89)) without setting its `random_state`. This means that the argument is set to its default `None` and therefore line [106](https://github.com/neuropsychology/NeuroKit/blob/8c564938c5fc2b4a377fd89bb14ac07a430983b9/neurokit2/signal/signal_distort.py#L106) resets the random generator to a random state.
A solution is to call `signal_distort` with 'random_state=np.random.randint(np.iinfo(np.uint32).max)`.
An even better solution, moving forward, would be to use a see generator object and pass it around from one function to all its callees, as suggested for example [here](https://albertcthomas.github.io/good-practices-random-number-generators/).
A simple way to do this is to replace the
```python
np.random.seed(random_state)
```
at the beginning of all functions and replace it with:
```python
rng = np.random.default_rng(random_state)
```
and use `rng` throughout the function (e.g. we need to replace `np.random.normal(0, 5, size=1)` with `rng.normal(0, 5, size=1)`) and pass it to all callees as `random_state=rng`. We also need to remove `np.random.seed(None)` from the end of each function (since we are not touching the singleton numpy random generator, there's no need to reset it to a random state).

lciti, 2023-02-17:
To maintain backward compatibility, one could generate the `rng` object as follows:
```python
rng = random_state if isinstance(random_state, (np.random.Generator, np.random.RandomState))
      else np.random.RandomState(random_state)
```
Basically `numpy` has two types of generators: Generator and RandomState. The former is where the new improvements are made while the latter has better reproducibility guarantees (see [here](https://numpy.org/doc/stable/reference/random/legacy.html#numpy.random.RandomState)). Using the line above, the person calling the function can choose between one of them by instantiating the corresponding object (possibly with a seed) ouside of the neurokit function and pass the object. Alternatively, if the user passes a seed (e.g. an int) or None, the more reproducible RandomState is used (which is what is used now under the hood, so code using a given seed now will give the same results even if/when this change is implemented).

lciti, 2023-02-18:
I can have a go later today and submit a PR

lciti, 2023-02-18:
@DominiqueMakowski I was having a look at it but there are some decisions to be made. The current code is a bit inconsistent and incorrect in the way the random state is propagated to callees (in addition to the bug above). For example `ppg_simulate` uses `random_state` to initialise the numpy singleton (global) generator. Eventually `_random_x_offset` is called internally and therefore it uses the generator initialised at the beginning of `ppg_simulate`. Later `signal_distort` is called with `random_state=random_state` which means that the generator inside `signal_distort` is initialised to the same seed already used to initialise `ppg_simulate`. Ideally one should not use the same seed twice but rather initialise a callee to a random state dependent on the current state in the caller (e.g. `random_state=np.random.randint(np.iinfo(np.uint32).max)`) or keep using the same generator of the caller in the callee (as I am proposing to do above). Doing the way it is done now means that, for example, the first Uniform number generated in `_random_x_offset` is not statistically independent of the first Laplace number generated inside `signal_distort`. For much later numbers I think things get more independent because generating numbers with different distributions may consume different numbers of draws from the underlying bit generators. Probably the issue is mostly theoretical and no results are at risk of being invalidated. Also, this problem only occurs when passing a seed, because otherwise the two `seed` functions get `None` and are initialised independently. Finally the numbers inside `_random_x_offset` are used in a different way than those in `signal_distort` (one is a time delay, one is an additive noise) so the dependency is unlikely to have major consequences.

For completeness, I want to show visually what I mean. Here is a scatter plot with `offsets[0]` generated inside [`_random_x_offset`](https://github.com/neuropsychology/NeuroKit/blob/8c564938c5fc2b4a377fd89bb14ac07a430983b9/neurokit2/ppg/ppg_simulate.py#L264) on the x axis and `_noise[0]` generated inside[`_signal_distort_noise`](https://github.com/neuropsychology/NeuroKit/blob/8c564938c5fc2b4a377fd89bb14ac07a430983b9/neurokit2/signal/signal_distort.py#L328) on the y axis over 1000 runs of `nk.ppg_simulate(random_state=random.randint(1, 1000000))`. If the two were independent, we would see a rectangular-ish cloud of points (i.e. the joint distribution of independent Uniform and Laplace RVs). Instead what we see is a deterministic dependency representing the mapping done inside numpy's C function [random_laplace](https://github.com/numpy/numpy/blob/b8feb4247af42cb0a27ddc7414d4a112e84d8d64/numpy/random/src/distributions/distributions.c#L469) to convert a Uniform number into a Laplace one (i.e. the same random Uniform number is returned as it is to `_random_x_offset` and converted to Laplace before being returned to `_signal_distort_noise`).
[image]

As I said, I think it is a minor thing. However if one decides to make the change that I recommend above, I think it is worth doing it the right way (i.e. passing the same generator (initialised once) to all callees). However data generated with a given seed from now on would not be exactly the same as data generated until now with that seed.
What do you think?
I think the options are:
- Break compatibility and do as I suggest in my second post above.
- Leave everything as it is and just "fix" `nk.eda_simulate` by adding `random_state=random_state` to make it behave like the other `xxx_simulate` functions.
- Do as I suggest in my second post above but also add a new argument to `xxx_simulate` called `distort_random_state` - the default (e.g. `pass` or 'propagate') means "do the right thing and propagate the generator of `xxx_simulate` to `signal_distort`" while any other value is passed to `signal_distort`'s `random_state` so that one can achieve backward compatibility by writing sth like `nk.ppg_simulate(random_state=1234, distort_random_state=1234)`.

If you let me know your preferred route I am happy to have a try.

DominiqueMakowski, 2023-02-19:
I am not super familiar with the random seeds, so I'd trust your call here. I think the best route is to break compatibility to get something clean and consistent, the 3rd option adds clunkiness to preserve a feature that I am not sure is super used anyway.

Thanks for giving it a go ☺️

lciti, 2023-02-19:
No problems. I'll giver it a try. One possible advantage of solution 3 is that incidentally it allows to simulate the same signal with different noises. For example one could call `nk.ppg_simulate(random_state=123, distort_random_state=456)` and `nk.ppg_simulate(random_state=123, distort_random_state=789)` and get the same simulated PPG "signal" after being distorted by two different noises. Of course one could always call  `nk.ppg_simulate(random_state=123)` and get a random (but consistent) distortion. I think I can try 3, from which we can easily make minor changes and go back to 1 if we think it's too clunky.

DominiqueMakowski, 2023-02-19:
Alright, makes sense. I'd just recommend the argument name to be `random_state_distort` (for consistency with other cases where we have for instance `method` and then `method_peaks`, `method_clean` etc)

PR #793 (2023-03-19): [Improvement] Fixes #784 and use random generator objects rather than np.random global RNG
# Description

This PR fixes #784 and make changes to the way random numbers are generated along the lines of what I describe in the discussion of #784.

# Proposed Changes

Modified the approach to random number generation (also solving the issue that the same seed was used both to generate a signal and its noise).
XXX_simulate functions now use an RNG object (RandomState or Generator) rather than the global singleton numpy.random one.
Their argument` random_state` can take several values, notably the f [... cut]
commit 9e5c2536 (2023-03-19): Merge pull request #793 from lciti/master

[Improvement] Fixes #784 and use random generator objects rather than np.random global RNG
commit b624ec1f (2023-03-19): Fixes #784 by having eda_simulate passing a random_state argument to signal_distort
closed 2023-03-19 by lciti