neuropsychology/NeuroKit #790: What's the source of VHF band (0.4-0.5Hz) of HRV sprecturm metrics

mobilemodem, 2023-02-23:
**Question and context**
I used the HRV toolbox from NeuroKit. and got a high correlation between VHF power and sleep-arousal quantity.
But I didn't find any document to describe the VHF band and its physiology meaning.
Hope someone help!

DominiqueMakowski, 2023-02-23:
The physiological meaning is not as straightforward as the mathematical meaning, and there is a vast literature discussing what could be the "sources" of high-frequency oscillations. Our team has written a bit on that that you might find helpful:

- https://www.mdpi.com/1424-8220/21/12/3998
- https://psyarxiv.com/mwa6x/

mobilemodem, 2023-04-26:
> Is this still relevant? If so, what is blocking it? Is there anything you can do to help move it forward?
> 
> This issue has been automatically marked as stale because it has not had recent activity. It will be closed if no further activity occurs.

I found some article mentioned that VHF index was related with failure of left vetricualr function.
https://ieeexplore.ieee.org/document/4745525.

closed 2023-08-07 by stale[bot]