neuropsychology/NeuroKit #794: Cancel cleaning pipeline

srulikbd, 2023-02-26:
Is there an option to cancel the cleaning pipeline, for an ECG processing?

Becasue, if I get  an already cleaned data, I think extra cleaning might be overprocessing..am I right?

Thanks :]

DominiqueMakowski, 2023-02-26:
In that case best is to construct your own [custom pipeline](https://neuropsychology.github.io/NeuroKit/examples/bio_custom/bio_custom.html)

That said @danibene, now that we start to have the nice `_methods()` functions, we could potentially allow something like `method_cleaning=None` which would make it skip the cleaning steps what to you think?

danibene, 2023-02-26:
@DominiqueMakowski agreed, like this?
https://github.com/danibene/NeuroKit/blob/3152652eb68b94a4bffff831067a7e3199af4586/neurokit2/eda/eda_methods.py#L96-L97

DominiqueMakowski, 2023-02-26:
Yes, but them make sure it is skipped by skipping the call to cleaning in eda_process Or, in `eda_cleaning()` have a sanity check that if method is none return the signal untouched. So that we do the "skipping" directly inside the functions, rather than in the upper layer? Not sure what's the best way tho

srulikbd, 2023-02-27:
Amazing!
Thanks, I'll check it up.

closed 2023-02-27 by srulikbd