neuropsychology/NeuroKit #801: Is the inhalation/exhalation detection in the respiratory signal consistent with the simulation?

thomastkok, 2023-03-07:
I am submitting this as a question, as I am not sure if this is correct and my mistake or an actual inconsistency. If I should resubmit it as a bug, please let me know.

The `rsp_simulate` function uses the BreathMetrics paper (Noto et al., 2018) as a source: https://academic.oup.com/chemse/article/43/8/583/5050471. Inspecting the signals in Figure 1 at the top of page 585 (page 3 of the pdf) indicates positions of inhalation onset, inhalation peak, exhalation onset, etc.

In the example of the `rsp_process` function, at https://neuropsychology.github.io/NeuroKit/functions/rsp.html and https://github.com/neuropsychology/NeuroKit, it can be seen that the inhalation and exhalation onsets are not consistent with the inhalation and exhalation onsets as noted in the paper. Instead, the inhalation and exhalation onsets are consistent with the inhalation and exhalation _peaks_ in the paper.

My guess is that this is either a mistake in the implementation of the peak detection for `rsp_process`, a misunderstanding of the simulated data by the BreathMetrics paper, or a misunderstanding by myself.

I think if needed, it can be adapted easily to apply the existing method to the numerical integration of the respiratory signal.

Thanks for taking the time to take a look at my question, and for developing such a useful library.

DominiqueMakowski, 2023-03-08:
That's an interesting observation, unfortunately it's been a long time and I don't really remember much of the RSP simulation. It is totally possible that an error has sneaked in somewhere. Would you mind helping us to get to the bottom of that by:

- trying with another peak detection method to see whether the default one used in `rsp_process()` is the suspect
- if peak detection works normally, trying to see where lies the problem in the implementation of the simulation

Thanks a lot ☺️ the RSP simulator is a fairly old piece of code, so I am sure it would benefit of having a fresh set of eyes going through it (feel free to open a PR if you feel like there are other adjustments/improvements that could be added)

thomastkok, 2023-03-08:
> That's an interesting observation, unfortunately it's been a long time and I don't really remember much of the RSP simulation. It is totally possible that an error has sneaked in somewhere. Would [...]

I actually think that both the peak detection and simulation work fine; but they operate on a different respiratory signal: `rsp_simulate()` generates an airflow-like signal, while using peak detection for inhalation/exhalation onset is appropriate for respiratory volume, but not for respiratory airflow (as far as my knowledge goes). I think the problem/solution lies in a consistent definition of what the respiratory signal is: airflow or volume.

As `rsp_process` also calculates the Respiratory Volume based on the signal, I assume airflow is more appropriate / the original intent. Then, the peak detection should just be applied to this RVT signal instead of the original signal. 

Alternatively, maybe the RSP could be more flexible in being either airflow or volume, but I suppose that would be more confusing. 

> Thanks a lot relaxed the RSP simulator is a fairly old piece of code, so I am sure it would benefit of having a fresh set of eyes going through it (feel free to open a PR if you feel like there are other adjustments/improvements that could be added)

If you agree with my conclusions, I can see to implement it and create a PR.

DominiqueMakowski, 2023-03-08:
> using peak detection for inhalation/exhalation onset is appropriate for respiratory volume, but not for respiratory airflow (as far as my knowledge goes)

Oh really? I would have naively expected the other way 🙃 

Yes there might be some confusion here as we are treat indistinctively different types of RSP signals but there are actually different types of them. 

I think by default we should make so that the RSP processing pipeline expects data from like a typical [respiratory belt](https://www.biopac.com/product/respiratory-effort-transducer-bsl/), just because that's proooobably the most common type of signals for RSP (at least in my field). And then we could have an argument like `type` that would modulate the type of processing based on the type of signals. 

Do you think that makes sense?

thomastkok, 2023-03-09:
> > using peak detection for inhalation/exhalation onset is appropriate for respiratory volume, but not for respiratory airflow (as far as my knowledge goes)
> 
> Oh really? I would have naively expected the other way upside_down_face
> 
> Yes there might be some confusion here as we are treat indistinctively different types of RSP signals but there are actually different types of them.
> 
> I think by default we should make so that the RSP processing pipeline expects data from like a typical [respiratory belt](https://www.biopac.com/product/respiratory-effort-transducer-bsl/), just because that's proooobably the most common type of signals for RSP (at least in my field). And then we could have an argument like `type` that would modulate the type of processing based on the type of signals.
> 
> Do you think that makes sense?

Just to confirm, I checked the source code of `rsp_simulate`, which calculates inhalation and exhalation onsets. Plotting that on the simulated signal gives this:

[image]

I do think that solution makes sense, and I agree that that should be the default signal (which also corresponds with the simulated signal). Should not be too much work to implement.

closed 2023-06-10 by stale[bot]