neuropsychology/NeuroKit #802: Couldn't install neurokit2 on kaggle.

Przemekeke, 2023-03-18:
Any ideas? I am getting this response using kaggle notebooks to while trying to install neurokit2:

```
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f6b61217650>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/neurokit2/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f6b61217c90>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/neurokit2/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f6b61217d10>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/neurokit2/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f6b61217150>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/neurokit2/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f6b611e40d0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/neurokit2/
ERROR: Could not find a version that satisfies the requirement neurokit2 (from versions: none)
ERROR: No matching distribution found for neurokit2
WARNING: There was an error checking the latest version of pip.
Note: you may need to restart the kernel to use updated packages.
```

Maybe it is to big package or it has to much dependencies. Maybe some lightweight version could be considered? 

Thanks

DominiqueMakowski, 2023-03-19:
The package depends on fairly standard dependencies (numpy, scipy, scikit-learn), all of which are often used on kaggle from my knowledge. I am not sure what's the cause of the error, as the error message is a bit obscure... really not sure we can fix that on our side 😟

Przemekeke, 2023-03-20:
Yes, I think it should be possible ofc, but the limitations of the kaggle virtual envs seems pretty high and installing neurokit might ends up in timeout. Anyhow thanks for quick reply