neuropsychology/NeuroKit #804: ecg_delineate miss the first P peak or last T peak

HotDog702, 2023-03-21:
nk version 0.2.3
I think, it is important, that lengths of P-peaks and T-peaks arrays were equal to length of R-peaks array, otherwise it is hard to group points of certain cardio cycle. But this is not the case, and often first P-peak or last T-peak is missed. In my code I've added a special workaround to this:
```
        for key in ['ECG_P_Peaks', 'ECG_P_Onsets', 'ECG_P_Offsets']:
            if ((len(waves[key]) > 0) and
                (len(waves[key]) < len(R_peaks)) and
                (waves[key][0] > R_peaks[0])):
                # lost first P-point
                waves[key].insert(0, np.nan)
        for key in ['ECG_T_Peaks', 'ECG_T_Onsets', 'ECG_T_Offsets']:
            if ((len(waves[key]) > 0) and
                (len(waves[key]) < len(R_peaks)) and
                (waves[key][-1] < R_peaks[-1])):
                # lost last T-point
                waves[key].append(np.nan)
```
Can you add this to the function?

danibene, 2023-03-24:
> Could you check with the dev version? I think some improvements were made recently but not sure if this exactly @danibene

I think you're referring to https://github.com/neuropsychology/NeuroKit/pull/777 , which prevented detection of points that weren't in the signal, but didn't ensure that the length of the arrays was the same

> Can you add this to the function?

I think adding this change to make it easier to group the points seems reasonable, but shouldn't we also do the same for Q and S points for consistency?

HotDog702, 2023-03-24:
> shouldn't we also do the same for Q and S points for consistency?

Yes, all arrays must be of the same length, as you already mark the invalid points with np.nan

danibene, 2023-03-24:
Would you like to make a PR ensuring that all points are the same length? Otherwise I can, probably in around a week

HotDog702, 2023-03-27:
Please, do this. Thank you

danibene, 2023-03-30:
> > shouldn't we also do the same for Q and S points for consistency?
> 
> Yes, **all** arrays must be of the same length, as you already mark the invalid points with np.nan

Do you have an example of invalid points being marked with np.nan in the output? I'm looking at the code now, and it seems to me that all NaNs are removed before returning the `waves` dictionary:

https://github.com/neuropsychology/NeuroKit/blob/408da03eb1c8d30093866387bdd31c2a7ab0dd96/neurokit2/ecg/ecg_delineate.py#L172-L175

I didn't write this code so I don't know what the intention was there but I'm unsure if it would make sense to add NaNs back in, as it isn't an error but more of a preference for how the output is formatted? Any thoughts @DominiqueMakowski ?

DominiqueMakowski, 2023-03-31:
@Tam-Pham

HotDog702, 2023-03-31:
Yes, there may be different thoughts, but my opinion is that amount of points must be equal, to easily assign its to cardiocycle.
As an option: totally change the output and group points by cycle, may be return list of dict

closed 2023-06-17 by stale[bot]