neuropsychology/NeuroKit #809: ecg_delineate mixes T and P peaks

HotDog702, 2023-03-27:
If P-peak is not distinct enough, the algorithm erroneously assign it position to previous T-peak
[image]

HotDog702, 2023-03-30:
Another example of P and R overlapping:
[image]
In ECG there are may be several P, R or T peaks, but their bounds must not overlap

danibene, 2023-03-30:
Would it be possible for you to provide the code and data to generate a minimal example?

DominiqueMakowski, 2023-03-31:
@HotDog702 that's not gonna be easy to solve, as it's not just a fix in the current algorithm but would require implementing a new and better one. 

If you're interested, I would invite you to do a quick literature search of the state of the art or ECG delineation algorithm, and see what implementations are available open-source. We can then discuss and potentially integrate some into neurokit.

HotDog702, 2023-05-22:
Don't quite understand: what is the correct (or at least, expected) order of delineated points?
Pon - P - Poff - Ron - Q - R - S - Roff - Ton -T -Toff , yes ? Could you plot the expected positions on some "ideal" ECG?
Are Ron & Roff correct names? May be Qon & Soff ?

closed 2023-09-17 by stale[bot]