neuropsychology/NeuroKit #818: ecg_peaks finding every other R peak

giodesp, 2023-05-06:
Hi all,
Firstly, thank you for Neurokit, what a powerful, excellently commented and easy-to-use piece of kit! 🙏
I have some rather noisy ECG data recorded during daylong home recordings. I've been testing the different cleaning and peak detection methods on some of the cleaner segments of my data and found a strange systematic detection of every _other_ R peak (sometimes it skips more than one peak, but it mostly skips over one peak every time):
[image]
The methods here are Pantompkins, Hamilton, and Gamboa; the other methods did not work well for me. 
If anyone has an idea why this is happening, I would appreciate you letting me know. I have manually coded the peaks and the BPM in this segment is 60 so it definitely shouldn't be within range of any cutoff filter.

Thank you!
Gio

DominiqueMakowski, 2023-05-07:
🤔 odd indeed, can you share a segment of data on which it fails?

giodesp, 2023-05-08:
Hi Dominique, sure, here is the ECG segment I was testing. To be clear, when I was applying, say, the pantompkins method, I used the associated ecg_clean and then ecg_peaks for that method.
[ecg.npy.zip](https://github.com/neuropsychology/NeuroKit/files/11419621/ecg.npy.zip)

danibene, 2023-05-12:
Hi @giodesp would it be possible for you to include the code that you used? The file you included doesn't have the sampling rate specified. 

I just guessed the sampling rate and tried applying the pantompkins method and no peaks were skipped in the first 10 seconds of the file you included: https://gist.github.com/danibene/dd4d62b6bd0217779a86d4ab922e3eaa
[image]

giodesp, 2023-05-12:
Hi danibene, 
Thanks so much for taking a look! You were right to

closed 2023-05-12 by giodesp

danibene, 2023-05-12:
Glad I could help @giodesp 😄

DominiqueMakowski, 2023-05-12:
Nice one Dani 

[image]