neuropsychology/NeuroKit #820: Error when detecting PPG peaks with Bishop's method

jatervon, 2023-05-11:
Hi,

I noticed that using the Bishop's method to detect peaks from a PPG/BVP signal ends up with a KeyError.

To verify and provide a reproducable example, I made a fresh install with

```
conda create -n nk_test python=3.9
conda activate nk_test
pip install pip setuptools wheel --upgrade
pip install neurokit2
```

Now `nk.version()` outputs

- OS: Linux (ELF 64bit)
- Python: 3.9.16
- NeuroKit2: 0.2.4

- NumPy: 1.24.3
- Pandas: 2.0.1
- SciPy: 1.10.1
- sklearn: 1.2.2
- matplotlib: 3.7.1

And then in Python

```
import neurokit2 as nk
ppg = nk.ppg_simulate(60, 64)
ppg_s, ppg_i = nk.ppg_process(ppg, 64, method_peaks='bishop')
```

returns the following error

Traceback (most recent call last):
  File "/home/xxx/miniconda3/envs/nk_test/lib/python3.9/site-packages/pandas/core/series.py", line 1159, in __setitem__
    self._set_with_engine(key, value)
  File "/home/xxx/miniconda3/envs/nk_test/lib/python3.9/site-packages/pandas/core/series.py", line 1222, in _set_with_engine
    loc = self.index.get_loc(key)
  File "/home/xxx/miniconda3/envs/nk_test/lib/python3.9/site-packages/pandas/core/indexes/range.py", line 350, in get_loc
    self._check_indexing_error(key)
  File "/home/xxx/miniconda3/envs/nk_test/lib/python3.9/site-packages/pandas/core/indexes/base.py", line 5736, in _check_indexing_error
    raise InvalidIndexError(key)
pandas.errors.InvalidIndexError: [[  67  120  183  241  304  362  424  482  539  599  653  708  761  810
   860  910  959 1010 1061 1111 1163 1216 1268 1327 1385 1443 1501 1559
  1623 1681 1741 1802 1856 1915 1967 2019 2072 2126 2173 2222 2275 2323
  2376 2424 2480 2533 2586 2643 2701 2759 2821 2881 2941 2999 3064 3120
  3176 3231 3284 3337 3390 3440 3489 3538 3589 3639 3690 3739 3792]]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/xxx/miniconda3/envs/nk_test/lib/python3.9/site-packages/neurokit2/ppg/ppg_process.py", line 90, in ppg_process
    peaks_signal = _signal_from_indices(info["PPG_Peaks"], desired_length=len(ppg_cleaned))
  File "/home/xxx/miniconda3/envs/nk_test/lib/python3.9/site-packages/neurokit2/signal/signal_formatpeaks.py", line 75, in _signal_from_indices
    signal[indices] = value
  File "/home/xxx/miniconda3/envs/nk_test/lib/python3.9/site-packages/pandas/core/series.py", line 1216, in __setitem__
    self._set_with(key, value)
  File "/home/xxx/miniconda3/envs/nk_test/lib/python3.9/site-packages/pandas/core/series.py", line 1238, in _set_with
    self._set_labels(key, value)
  File "/home/xxx/miniconda3/envs/nk_test/lib/python3.9/site-packages/pandas/core/series.py", line 1255, in _set_labels
    raise KeyError(f"{key[mask]} not in index")
KeyError: '[(67, 120, 183, 241, 304, 362, 424, 482, 539, 599, 653, 708, 761, 810, 860, 910, 959, 1010, 1061, 1111, 1163, 1216, 1268, 1327, 1385, 1443, 1501, 1559, 1623, 1681, 1741, 1802, 1856, 1915, 1967, 2019, 2072, 2126, 2173, 2222, 2275, 2323, 2376, 2424, 2480, 2533, 2586, 2643, 2701, 2759, 2821, 2881, 2941, 2999, 3064, 3120, 3176, 3231, 3284, 3337, 3390, 3440, 3489, 3538, 3589, 3639, 3690, 3739, 3792)] not in index'

which seems to indicate that peaks are returned in incorrect format.

Indeed, adding `peaks = peaks[0]` to line 233 in `ppg_findpeaks.py` worked like a charm. 

Is there are a reason why peaks are returned as a 2D array or should the change be made?

DominiqueMakowski, 2023-05-11:
Linking @peterhcharlton here as I think you helped adding that method if I remember?

Pi-Wi, 2023-05-22:
I also have a problem when using the ppg_process method:
`signals, info = nk.ppg_process(ppg_clean, sampling_rate=sample_rate_ppg, method="bishop")`
gives a ValueError:
`ValueError: `method` not found. Must be one of 'elgendi' or 'nabian2018'.`
When I then use it with the suggested keyword "nabian2018" I still get a ValueError:
`ValueError: `method` not found. Must be one of the following: 'elgendi', 'bishop'.`

Seems also in the docstring of the method there is something wrong/missing:
`    method : str
        The processing pipeline to apply. Can be one of ``"elgendi"``.
        Defaults to ``"elgendi"``.`

Not sure if this should be a new issue or within the same.

danibene, 2023-05-30:
> I also have a problem when using the ppg_process method: `signals, info = nk.ppg_process(ppg_clean, sampling_rate=sample_rate_ppg, method="bishop")` gives a ValueError: `ValueError: 'method' not [...]

Hi @Pi-Wi I think this is because `bishop` does not have an associated method to clean the signal, so it's not an option as a complete processing pipeline method. To use the peaks method with another cleaning method, you can run e.g.
```python
signals, info = nk.ppg_process(ppg, sampling_rate=sampling_rate, method_cleaning="elgendi", method_peaks="bishop")
```

@DominiqueMakowski I think the issue originally brought up by @jatervon is the same issue as https://github.com/neuropsychology/NeuroKit/issues/831 so once we close that one we should be able to close this too?

closed 2023-05-30 by danibene