neuropsychology/NeuroKit #823: ecg_peaks (method='neurokit') sometimes fails on ECG's with missing data

tecamenz, 2023-05-12:
**Describe the bug**
On certain ECG's with missing data or longer periods of constant values (e.g. zeros), `ecg_peaks` with `method="neurokit"` fails to find all but two peaks. Below I attached a screenshot with `show=True`:
[image]

The cause of this issue lies in the way Neurokit calculates the `min_len`. This variable defines the minimum qrs length (`len_qrs`) required to accept a valid R-peak. 
https://github.com/neuropsychology/NeuroKit/blob/3d5ecfca8e9c10a1f4150f8877aba8f34c0d0a22/neurokit2/ecg/ecg_findpeaks.py#L249-L252

Changing `np.mean()` to `np.median` would provide a more robust estimate for the most common qrs length and would prevent `min_len` from becoming too high. The following screenshot shows the detected R-peaks with `np.median()`. 
Because the median calculation is a bit slower, I saw a 16% increase in runtime on a 24h ECG. 

[image]

**System Specifications**
- OS: Linux ( 64bit) 
- Python: 3.8.10 
- NeuroKit2: 0.2.0 

- NumPy: 1.23.4 
- Pandas: 1.5.3 
- SciPy: 1.10.1 
- sklearn: 1.2.2 
- matplotlib: 3.6.2

DominiqueMakowski, 2023-05-13:
Interesting. 

@JanCBrammer what do you think about changing allowing for median instead of mean? 

I reckon we should probably do some benchmarking and some tests to see whether the median option is as good as mean in most cases

JanCBrammer, 2023-05-15:
@DominiqueMakowski, exactly my thoughts.
It's an interesting and useful observation. The median looks promising and on paper it's preferable over the mean in this context. 
But it'd be good to confirm that (i.e., the medians "superiority") empirically.

JanCBrammer, 2023-05-15:
Do we have benchmarking set up?
Otherwise the [benchmarking over at biopeaks](https://jancbrammer.github.io/biopeaks/contributor_guide.html#algorithm-benchmarks) would do, since it's the [same peak detection](https://github.com/JanCBrammer/biopeaks/blob/81f3c8bbf7344ef171f6b17bff8c25f9d137b5df/biopeaks/heart.py#L77-L80).

DominiqueMakowski, 2023-05-15:
> Do we have benchmarking set up?

Not really (aside from [this old thread](https://github.com/neuropsychology/NeuroKit/issues/245)).

It seems like we've all at some point done bits of work on our side, it would be really cool to assemble around a "task force" project (also with @cbrnr as they did [some cool work](https://sleepecg.readthedocs.io/en/stable/heartbeat_detection/#benchmarks)) for a cross-software benchmarking initiative...

tecamenz, 2023-08-20:
Interesting approach. But then the question would be how to determine the optimal thresholds for flat-line detection. How do you define "flat" in the presence of noise or random artifacts (large voltage spikes). How does this approach compare to the simple median?

DominiqueMakowski, 2023-08-21:
Very cool indeed. I also had some issues recently with bad chunks of data, it would be good to have a robust method to either ignore them and/or adapt the parameters in them... 

However, I'm not sure what the best API for that should look like, because it's not only peak detection that should be adapted to handle bad intervals. @danibene already started to make some functions resistant to missing values (https://github.com/neuropsychology/NeuroKit/issues/838).

Maybe:

- `*_peaks()` would gain a `bad_intervals` argument (e.g., `bad_intervals=[(300, 450), (680-700)]`) with periods (that can be also entered manually) it would exclude/adapt. Importantly It would *return* these in the info dict.
- Upstream: in general `*_process()` functions, we could incorporate a step of bad_intervals detection that would then get piped into bad_intervals
- Downstream: other functions relying on peaks `*_rate()` etc. should have this argument as well and deal with them appropriately, for instance by setting the rate to NaNs on these bad intervals

HotDog702, 2023-08-22:
My thoughts about this topic:
1) Long term records are a-priori not stationary, so we should split it for several shorter chunks and process separately, then combine
2) From the above we can estimate thresholds from each chunk, then analyze regularity of found peaks, select the correct thresholds and re-process the chunks with bad results
3) As a help, we can also use the expected ECG values ~1 mV and HR ~30..300 BPM as a quick check of threshold correctness

closed 2025-02-01 by stale[bot]