neuropsychology/NeuroKit #838: Better handling of missing signal

DominiqueMakowski, 2023-06-06:
Signal interruption can happen, currently we pad the missing signal early on (during cleaning) and then proceed normally.

```python
import neurokit2 as nk

signal = np.concatenate(
    [
        nk.ecg_simulate(duration=10, sampling_rate=100),
        [np.nan] * 1000,
        nk.ecg_simulate(duration=10, sampling_rate=100),
    ]
)
nk.signal_plot(signal)

df, _ = nk.ecg_process(signal, sampling_rate=100)
df.plot(subplots=True, figsize=(10,10))
```
[image]

We should probably fill with NaNs at the indices where the raw signal is NaN. The only problem is that we need to make sure that our feature computation function (e.g., HRV, etc) are nan-proof, and return nan if nans are present (but do not error).

danibene, 2023-06-07:
> The only problem is that we need to make sure that our feature computation function (e.g., HRV, etc) are nan-proof, and return nan if nans are present (but do not error).

Would you say returning NaN whenever there are any NaNs present would be the desired behavior for all features, or just some? 

In https://github.com/neuropsychology/NeuroKit/pull/720 I tried adapting the HRV features for missing R-R intervals but some features required more adaptations than others (e.g. for the MeanNN we can just ignore the missing intervals whereas for RMSSD I removed differences detected as non-consecutive). 

I'm still not confident that I adapted all the features in the most "ideal" way (in terms of best approximating the value that would be obtained without missing data), and I imagine this is something we would have to determine empirically if we wanted to be sure.

But I guess a start could be to return NaN for all functions that would currently raise an error, and then slowly adapt these functions for the missing data if we think we can get a reasonable approximation?

DominiqueMakowski, 2023-06-07:
> But I guess a start could be to return NaN for all functions that would currently raise an error, and then slowly adapt these functions for the missing data if we think we can get a reasonable approximation?

I think that's good. And indeed I expect many cases / indices where dealing with missing data is not straightforward, in that case it's safer just to return NaNs and it will be up to the user to explicitly do something about it (analyze separately various chunks, input signal etc.)

closed 2023-09-17 by stale[bot]