neuropsychology/NeuroKit #844: ECG function error

undefinedMethod, 2023-06-18:
**Describe the bug**
When processing any ecg data I get the error 
ValueError: operands could not be broadcast together with shapes (n,) (n,)

Looks like something is causing the array that holds the number of peaks in the data to be passed with the array that holds the raw ecg data to the wrong function.

**error log** (the message related to the error).

**System Specifications**

It's important that you give us some information about the system you are using. For that you can run:

```python
import neurokit2 as nk
nk.version()
```
- OS: Linux (ELF 64bit) 
- Python: 3.10.12 
- NeuroKit2: 0.2.4 

- NumPy: 1.25.0 
- Pandas: 1.5.3 
- SciPy: 1.10.1 
- sklearn: 1.2.2 
- matplotlib: 3.7.1

DominiqueMakowski, 2023-06-18:
Do you have a reproducible example?

undefinedMethod, 2023-06-18:
If I run the example code;

`# Load NeuroKit and other useful packages
import neurokit2 as nk
import numpy as np
import matplotlib.pyplot as plt

# Note: this cell is hidden using the "remove-input" tag
# Make bigger images
plt.rcParams['figure.figsize'] = [15, 5]  
plt.rcParams['font.size']= 14`

`# Simulate 30 seconds of ECG Signal (recorded at 250 samples / second)
ecg_signal = nk.ecg_simulate(duration=30, sampling_rate=250)`

`# Automatically process the (raw) ECG signal
signals, info = nk.ecg_process(ecg_signal, sampling_rate=250)`

---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[3], line 2
      1 # Automatically process the (raw) ECG signal
----> 2 signals, info = nk.ecg_process(ecg_signal, sampling_rate=250)

File ~/.cache/pypoetry/virtualenvs/timing-threat-data-analysis-GrIVHu1n-py3.11/lib/python3.11/site-packages/neurokit2/ecg/ecg_process.py:97, in ecg_process(ecg_signal, sampling_rate, method)
     92 # R-peaks
     93 (instant_peaks, rpeaks,) = ecg_peaks(
     94     ecg_cleaned=ecg_cleaned, sampling_rate=sampling_rate, method=method, correct_artifacts=True
     95 )
---> 97 rate = signal_rate(rpeaks, sampling_rate=sampling_rate, desired_length=len(ecg_cleaned))
     99 quality = ecg_quality(ecg_cleaned, rpeaks=rpeaks["ECG_R_Peaks"], sampling_rate=sampling_rate)
    101 signals = pd.DataFrame(
    102     {"ECG_Raw": ecg_signal, "ECG_Clean": ecg_cleaned, "ECG_Rate": rate, "ECG_Quality": quality}
    103 )

File ~/.cache/pypoetry/virtualenvs/timing-threat-data-analysis-GrIVHu1n-py3.11/lib/python3.11/site-packages/neurokit2/signal/signal_rate.py:82, in signal_rate(peaks, sampling_rate, desired_length, interpolation_method)
      5 def signal_rate(
      6     peaks, sampling_rate=1000, desired_length=None, interpolation_method="monotone_cubic"
      7 ):
      8     """**Compute Signal Rate**
      9 
     10     Calculate signal rate (per minute) from a series of peaks. It is a general function that works
   (...)
     80 
     81     """
---> 82     period = signal_period(peaks, sampling_rate, desired_length, interpolation_method)
     83     rate = 60 / period
     85     return rate

File ~/.cache/pypoetry/virtualenvs/timing-threat-data-analysis-GrIVHu1n-py3.11/lib/python3.11/site-packages/neurokit2/signal/signal_period.py:86, in signal_period(peaks, sampling_rate, desired_length, interpolation_method)
     84 # Interpolate all statistics to desired length.
     85 if desired_length is not None:
---> 86     period = signal_interpolate(peaks, period, x_new=np.arange(desired_length), method=interpolation_method)
     88 return period

File ~/.cache/pypoetry/virtualenvs/timing-threat-data-analysis-GrIVHu1n-py3.11/lib/python3.11/site-packages/neurokit2/signal/signal_interpolate.py:103, in signal_interpolate(x_values, y_values, x_new, method, fill_value)
    100     x_new = np.linspace(x_values[0], x_values[-1], x_new)
    101 else:
    102     # if x_values is identical to x_new, no need for interpolation
--> 103     if np.all(x_values == x_new):
    104         return y_values
    106 # If only one value, return a constant signal

ValueError: operands could not be broadcast together with shapes (34,) (7500,)

undefinedMethod, 2023-06-18:
I don't think it is an issue with my jupyter lab setup as I get the same error when running the same code in a python script

gtzedaki, 2023-06-22:
I had the same issue with numpy 1.25 installed.
Downgrading to 1.24.3 indeed solves the issue.

DominiqueMakowski, 2023-06-22:
@danibene do you have any idea why it breaks on numpy 1.25 🤔

gtzedaki, 2023-06-22:
Unfortunately no, a dimension mismatch happens for some reason.

danibene, 2023-06-22:
@DominiqueMakowski I had the same issue! I think if we release 0.2.5 it should be fixed due to this commit:
https://github.com/neuropsychology/NeuroKit/pull/814/commits/ce73a2d155475b0f997f2cd27013c661fbf4152e

danibene, 2023-08-10:
Now that 0.2.5 has been released I am closing this

closed 2023-08-10 by danibene