neuropsychology/NeuroKit #847: Visualizing Identified Peaks for Respiration

melisasaygin, 2023-06-21:
I cleaned the mock respiration signal with the nk.rsp_clean function (khodadad 2018). I have been trying to find the peaks and then visualize it on top of the actual signal to make sure it is visually plausible. I have been trying with different datasets and methods (scipy, khodadad2018, etc.) but the graph I get always looks like the attached, and never on top of each other. Does anyone know how I can fix it? I split a big dataframe with all the tasks' into different data frames, each representing one task, which is why the signal starts from sample 145920 and not zero. I would really appreciate any help. I could also do it for the whole big dataset but in that case, I get a full red box because there are so many samples and lines, and in that case I do not know how I can zoom into that somehow.
[image]

melisasaygin, 2023-06-21:
This actually helped a lot! But still I cannot clearly see due to there being many waveforms in one graph. Do I have to keep partitioning the dataframe and make it start from 0, or is there a simpler way of just doing a quick zoom in (with square brackets, for ex.)? 
[image]

DominiqueMakowski, 2023-06-21:
> a simpler way of just doing a quick zoom in (with square brackets, for ex.)

Yes there is, need probably do to something like `data.iloc[10000:30000]` for example (or something like that)

danibene, 2023-06-22:
for an interactive plot, you could try passing `report="report.html"` to `rsp_process()` to generate an interactive plot (and use the `method_peaks` argument to specify the peak detection method)

or if you want just the interactive plot you could set `static=False` in `rsp_plot()`

closed 2023-10-15 by stale[bot]