neuropsychology/NeuroKit #850: Eog features

VladimirR46, 2023-06-23:
Good afternoon! When I try to use the eog_features function I get the following error.
```
eog_cleaned = nk.eog_clean(ica0, sampling_rate=1000, method='neurokit')
peaks = nk.eog_findpeaks(eog_cleaned, sampling_rate=1000)
info = nk.eog_features(eog_cleaned, peaks,sampling_rate=1000)
```
But if I change sampling_rate to 100, for example, it works. 

`info = nk.eog_features(eog_cleaned, peaks,sampling_rate=100)`

But I have a frequency of 1000 Hz.

eog_cleaned and peaks have the right data type

danibene, 2023-06-26:
Would it be possible for you to share some example data? I ran your code with other data and did not get the same error:
https://gist.github.com/danibene/595f281bc118541e49b7d4462633d8ad

VladimirR46, 2023-08-01:
Of course!
You can open the file as:

```
with open('EOG.npy', 'rb') as f:
    eog_v = np.load(f)
```

[EOG.zip](https://github.com/neuropsychology/NeuroKit/files/12232594/EOG.zip)

danibene, 2023-08-02:
> eog_cleaned = nk.eog_clean(ica0, sampling_rate=1000, method='neurokit')
> peaks = nk.eog_findpeaks(eog_cleaned, sampling_rate=1000)
> info = nk.eog_features(eog_cleaned, peaks,sampling_rate=1000)
> ``` [...]

Thanks @VladimirR46 ! Could you please clarify what you meant by a frequency of 1000 Hz if the sampling rate=500?

VladimirR46, 2023-08-02:
Hi. It's just another (new) dataset that has a sampling rate of 500 Hz. 
But here I get the same error message.

```
eog_cleaned = nk.eog_clean(eog_v, sampling_rate=500, method='mne')
peaks = nk.eog_findpeaks(eog_cleaned, method='neurokit', sampling_rate=500)
info = nk.eog_features(eog_cleaned, peaks,sampling_rate=500)
```

[image]

I also need to get the "positive amplitude velocity ratio (pAVR)" characterization as in "Robbins, K. A. (2017). BLINKER: automated extraction of ocular indices from EEG enabling large-scale analysis."

[image]

But the resulting values are very different, i.e. I mean the value on the X-axis.

mean (matlab) == 5.3034
mean (neurokit2) == 9715.09827

VladimirR46, 2023-08-03:
Hey, everybody! I think I've solved my problem. 
1) I set threshold=0.3 for the function nk.eog_findpeaks

```
eog_cleaned = nk.eog_clean(eog_v, sampling_rate=500, method='mne')
peaks = nk.eog_findpeaks(eog_cleaned, method='neurokit', threshold=0.3, sampling_rate=500)
info = nk.eog_features(eog_cleaned, peaks,sampling_rate=500)
```

2) I changed the formula for calculating pAVR
```
#pAVR = abs(change_close.max() / duration_close) * 100
pAVR = abs(blink_close.max() / change_close.max())/sampling_rate * 100
pAVR_list.append(pAVR)
```

[image]

VladimirR46, 2023-08-10:
Good day! @[danibene](https://github.com/danibene)

```
#pAVR = abs(change_close.max() / duration_close) * 100

# My Fix
pAVR = abs(blink_close.max() / change_close.max())/sampling_rate * 100  
```

  Could you explain why the equation doesn't match the original article?

danibene, 2023-08-10:
Hi @VladimirR46 thanks for sharing your solution and error, it's very appreciated! 

I'm not the person who originally wrote this code and am not familiar with the article so it's hard for me to say off the top of my head (and I don't have time to look into it currently). 

@DominiqueMakowski are you more familiar with this?

In any case I would suggest we keep the issue open so that we eventually either change the code for calculation or give a more informative error message.

VladimirR46, 2023-10-16:
Yes, this question is still relevant.

closed 2025-02-24 by stale[bot]