neuropsychology/NeuroKit #851: Questions on EDA processing with Neurokit2

yingquehku, 2023-06-26:
Dear developers,

Hello, this is Ying. I and our team member are using neurokit2 to find SCR features (i.e., EDA peaks). We are having some questions when processing the data. Grateful if you can take a look when you are available. Thanks a lot!

Our EDA signals were collected by the Empatica E4 wristband and the sampling rate is low (4Hz). We found a custom processing pipeline for the data from Empatica provided by isayasMatter in this answer: https://github.com/neuropsychology/NeuroKit/issues/554
[image]

1. We found you formally provided a function named eda_process() which could return the values of EDA_phasic, SCR_Peaks. Our question is that if it is similar to the output from the aforementioned eda_custom_process which used nk.eda_phasic, eda_peaks. (https://neuropsychology.github.io/NeuroKit/functions/eda.html#main)

2. We observed that in this formal page: https://neuropsychology.github.io/NeuroKit/functions/eda.html#main, the sampling rate in the function of eda_process() indicates "sampling_rate (int) – The sampling frequency of "rsp_signal" (in Hz, i.e., samples/second)". We are not sure if it is a typo that the sampling frequency is equal to rsp_signal instead of the EDA signals.

4. We are wondering if there are any referenced ranges for the scr_peaks, scr_amplitude, etc. for a certain period of time.

5. We observed "nan" outputs in SCR_Amplitude, SCR_RiseTime, and SCR_RecoveryTime. May I ask what these nan values mean? As we hope to further calculate SCR_Mean_Amplitude, SCR_Max_Amplitude, SCR_Mean_RiseTime, SCR_Mean_RecoveryTime from the above outputs, we propose to remove these "nan" values before further calculation. Does this make sense?

Looking forward to your reply and thank you very much!

Best,
Ying

danibene, 2023-06-26:
I will try to answer some of your questions:
 
> 1. We found you formally provided a function named eda_process() which could return the values of EDA_phasic, SCR_Peaks (https://neuropsychology.github.io/NeuroKit/functions/eda.html#main). Our question is that if it is similar to the output from the aforementioned eda_custom_process which used nk.eda_phasic, nk.eda_peaks.

`eda_process()` also includes a function to clean the data, whereas the `eda_custom_process()` function you showed does not. This custom pipeline was originally written because the sampling rate of the empatica is too low for the filtering applied in the default cleaning function. 

However, `eda_process()` has been since updated to skip the filtering if the sampling rate is too low, so you should get the same output as `eda_custom_process()`, with a warning:
https://github.com/neuropsychology/NeuroKit/blob/35ed696300626eadd9ae953875556c64ccca9190/neurokit2/eda/eda_clean.py#L104-L108

> 2. We observed that in this formal page: https://neuropsychology.github.io/NeuroKit/functions/eda.html#main, the sampling rate in the function of eda_process() indicates "sampling_rate (int) – The [...]

Thank you for pointing that out! It's a typo, yes.

> 4. We observed "nan" outputs in SCR_Amplitude, SCR_RiseTime, and SCR_RecoveryTime. May I ask what these nan values mean?

Are these nan values only at the beginning, before the first onset? If so, I think that is to be expected. Feel free to also provide a minimal example with sample data if you'd like to show exactly what your output looks like.

DominiqueMakowski, 2023-06-27:
And 
3. Not to my knowledge :)
1. If you install the dev version of neurokit, you can use  `eda_process()` directly on your Empatica data (no need to build a custom pipeline)

yingquehku, 2023-06-27:
Dear [DominiqueMakowski](https://github.com/DominiqueMakowski), 

Thanks for your quick response and answers.

Dear [danibene](https://github.com/danibene),
> Are these nan values only at the beginning, before the first onset? If so, I think that is to be expected. Feel free to also provide a minimal example with sample data if you'd like to show exactly what your output looks like.

Yes, we found there were few nan values at the very beginning of SCR_Onsets and SCR_Amplitude in some sample data (not all sample data). However, it was the most case that there were much more nan values in SCR_Recovery and SCR_RecoveryTime. May I know if is there any explanation for this? Thanks!

The below screenshots were SCR outputs from one sample data, including a couple of nan values at the beginning of SCR_Onsets, SCR_Amplitude, SCR_RiseTime, but more nan values in SCR_Recovery, and SCR_RecoveryTime.
[image]
[image]
[image]

The below screenshots were SCR outputs from another sample data, including nan values in SCR_Recovery and SCR_RecoveryTime.
[image]
[image]
[image]
[image]

danibene, 2023-06-27:
@yingquehku would it be possible for you to upload the data directly as well as the code tou used?

yingquehku, 2023-06-27:
> @yingquehku would it be possible for you to upload the data directly as well as the code tou used?
Sure. I updated the sample data as well as the code in the below attachment. 
Thank you very much!
[scr nan_20230627.zip](https://github.com/neuropsychology/NeuroKit/files/11883216/scr.nan_20230627.zip)

danibene, 2023-06-28:
@yingquehku thank you! I'll try to have a look at it sometime during the next week or two (unless figures it out before me), feel free to @ me if I don't get back to you by then

yingquehku, 2023-06-28:
> @yingquehku thank you! I'll try to have a look at it sometime during the next week or two (unless someone else figures it out before me), feel free to @ me if I don't get back to you by then

alright, thanks!

DominiqueMakowski, 2023-06-28:
Can you add a plot of your signal? In particular in areas when recovery time is missing?

yingquehku, 2023-06-29:
Thanks Dom for the advice! I followed this instruction (https://neuropsychology.github.io/NeuroKit/examples/eda_peaks/eda_peaks.html) to visualize the SCR features including SCR_Onsets, SCR_Peaks, SCR_Recovery in cleaned EDA signals for 8 passage reading periods of a participant. 

We plotted SCR_Onsets='red', SCR_Peaks='blue', SCR_Recovery='orange'.
From my observation, there were dense lines in the plot with missing recovery time, as the below first figure shows. In contrast, there were sparse lines in the plot without missing recovery time, as the below second figure shows.
[image]
[image]

I saved the SCR features outputs as well as the plots in this document: [plot scr features_20230628.docx](https://github.com/neuropsychology/NeuroKit/files/11901143/plot.scr.features_20230628.docx). 
Attached is the code and sample data I used for further reference.
[scr nan_20230628.zip](https://github.com/neuropsychology/NeuroKit/files/11901160/scr.nan_20230628.zip)

DominiqueMakowski, 2023-06-29:
It seems like you need first to apply some filtering + signal decomposition to your signal before estimating peaks (which should be done on the phasic component). Here the high-frequency noise messes up with the detection

yingyque, 2023-07-01:
Thanks Dom! I first used cvxEDA in nk.eda_phasic to pre-process the EDA data shown in the above dense-signal figure. The results showed that the data after cvxEDA processing look more reasonable, which decreases the high-frequency noises. 

cvxEDA
[image]

For the filtering, I tested the "smoothmedian" and the "highpass" embedded in the nk.eda_phasic function, but they seem to less work to remove the high-frequency noises in this case. My question is, as the cvxEDA seems good to solve this situation, do we need to **additionally use another method** to filter the high-frequency noises apart from using cvxEDA? 

smoothmedian
[image]

highpass
[image]

DominiqueMakowski, 2023-07-01:
[image]

This image looks like it's the raw data, it has 1) clear phasic peaks (SCRs) and 2) an overall negative drift (which is common), part of the tonic activity. 

Applying eda_phasic extracts the phasic (and the tonic) components: what you should focus on then is the phasic one, which should be on average **flat** (i.e., without the negative trend) and only show the peaks. From your pictures below it seems like you are working with the tonic signal, which should only have a negative trend without peaks (as in the pics)

closed 2023-10-15 by stale[bot]