neuropsychology/NeuroKit #857: Getting the Inspiratory and Expiratory Amplitudes Separately

melisasaygin, 2023-07-03:
**Question and context**

I did cleaning, processing, and interval-related feature extraction for a respiration signal. Although the inspiratory and expiratory times are very useful, I still need to separately get the amplitudes (depth) for inspiratory and expiratory periods. I know that the peaks are identified correctly. Is there a way to also get those features instead of just the average overall amplitude?

DominiqueMakowski, 2023-07-14:
Sorry for the late reply. Once you ran `rsp_process()` normally you have a continuous Amplitude signal, you can take just take its value at the peaks. 

Otherwise, you can manually compute the raw signal's values at peaks and troughs, and then subtract them to get all the amplitudes for each cycles.

hope that helps

melisasaygin, 2023-07-14:
What is the continuous amplitude relative to? It says that for peaks, the standard method gives it relative to the previous trough. So for the troughs, do I also make it relative to the previous trough? (By the way, this would be a very nice feature to add to the rsp toolkit, in my opinion so that inspiratory and expiratory flow rate can be directly calculated)

closed 2023-10-15 by stale[bot]