neuropsychology/NeuroKit #864: Correct Peak Detection but Unexpected Interval Features - Respiration

melisasaygin, 2023-07-12:
I would extremely appreciate your help and input.
I got data from a wearable that gives some signals with 256 Hz (ECG) sampling and some as 128 Hz (respiration, the focus of this question) all in the same raw data frame. So the respiration and ECG signals have the same number of rows (though I did time-matching so I know which intervals are which tasks). I processed the whole 1-hour respiration signal with the default method (as it worked very well) and then separated into different tasks so I can get the interval related features per task (every task is 30 sec). The attached is an example task called 12a. Everything is fine, except, when I try to get the interval related features and enter 128 Hz as my sampling rates, it calculates everything as if the time of the task is longer (I think it is because interval function thinks the task is 60 seconds and not 30 seconds due to there being more data points because of the fact that in the very beginning it was in the same dataframe with Hexoskin). You can see that the features it gives are wrong by looking at how many cycles of breathing there are, knowing the task is 30 seconds, and seeing the inspiration and expiration durations it gives (adds up to 60 seconds). Can this be easily fixed if I just enter 256 as my sampling rate (although respiration itself was sampled at 128)? Or what else can I do?

[image]
[image]
[image]
[image]

DominiqueMakowski, 2023-07-12:
How can a recording of the same length have the 2 signals with the same number of data points if the sampling rate is different? I may have missed something here

melisasaygin, 2023-07-12:
Hexoskin simultaneously records ECG (256 Hz) and Respiration (128 Hz) and gives the raw data in edf format all signals in the same table, having the same number of rows. 

By the way, when I left out the sampling_rate input from the interval related features, it did halve the values. (Comparable to what I initially did which was to separately process and calculate features per task, which was not a good enough approach). But I do not really understand the logic of this and how it functions better without the sampling rate input (which I did of course put when I was processing the whole 1 hour respiration data).

DominiqueMakowski, 2023-07-12:
> the raw data in edf format all signals in the same table, having the same number of rows.

But does that mean that the values of respiration are duplicated? Or upsampled to the max sampling rate? Or is that signal for instance shorter and padded with nan at the end?

melisasaygin, 2023-07-12:
I used mne package to read the edf into Jüpiter Notebook and it does the following: "  Another approach, which is the one taken by both (FieldTrip)[https://github.com/fieldtrip/fieldtrip/blob/master/edf2fieldtrip.m] and (mne)[https://github.com/mne-tools/mne-python/blob/master/mne/io/edf/edf.py] is to resample the channels to the highest sampling rate used in the EDF file, so that the data can be returned in a matrix  "

Does this mean I need to re-run everything always entering 256 Hz instead of 128 for respiration although that was not the genuine sampling? From what I did so far, all my visualizations, peaks, troughs, interval related features look okay (but if inputting 128, the genuine sampling was wrong, I will re-run of course).

DominiqueMakowski, 2023-07-12:
> Does this mean I need to re-run everything always entering 256 Hz

Yes, it's mostly so that the algorithms can know how much time separates 2 samples

closed 2023-10-15 by stale[bot]